erlerobot/gym-gazebo #87: apm_sitl.launch

jianhuaixie, 2018-01-21:
while processing /home/xxx/catkin_ws/src/mavros/mavros/launch/apm_sitl.launch:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/jianhuaixie/catkin_ws/src/mavros/mavros/launch/apm_sitl.launch'﻿

jianhuaixie, 2018-01-21:
when i run the code : python erlecopter_hover_qlearn.py
I get the error like this:
Another terminal show this: Can't execvp /home/xxx/ardupilot/Tools/autotest/sim_vehicle.sh
No such file or directory
while processing /home/xxx/catkin_ws/src/mavros/mavros/launch/apm_sitl.launch:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/jianhuaixie/catkin_ws/src/mavros/mavros/launch/apm_sitl.launch'﻿

I dowload a script named sim_vehicle.sh in the old version of project in github.I don't know it is working?

vmayoral, 2018-02-18:
This isn't a supported environment anymore. Refer to https://github.com/erlerobot/gym-gazebo#community-maintained-environments for those supported.

closed 2018-02-18 by vmayoral