erlerobot/gym-gazebo #90: Complete Installation notes for Python3, Ubuntu 16.04

zishanahmed08, 2018-01-23:
This is not an issue ,but  installation notes  Python3, Ubuntu 16.04 .Some of the errors may not be applicable for you.

**create virtual conda env**
`conda create -n p3 python=3.5.2 anaconda`

here onwards the installation continues in the conda environment created

**ROS installation** 
```
sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116
sudo apt-get install ros-kinetic-desktop
sudo rosdep init
rosdep update
echo "source /opt/ros/kinetic/setup.bash" >> ~/.bashrc
source ~/.bashrc
sudo apt remove '.*gazebo.*' '.*sdformat.*' '.*ignition-.*'
```

**GAZEBO8 installation**
```
sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
sudo apt-get update
sudo apt-get install gazebo8
```

**reboot might be required if gazebo not launching**
`sudo apt-get install ros-kinetic-gazebo8-ros-pkgs ros-kinetic-gazebo8-ros-control`

**rename cv2.so to another name(cv2-renamed.so) at /opt/ros/kinetic/lib/python2.7/dist-packages/**

```
pip install opencv-python
pip install gym
```

**For solving the boost package error related to c make refer this  [post](https://github.com/BVLC/caffe/issues/4843)**
```
cd /usr/lib/x86_64-linux-gnu
sudo ln -s libboost_python-py35.so libboost_python3.so 
```
`pip install rospkg catkin_pkg`

`sudo apt-get install python3-pyqt4`

```
sudo apt-get install \
cmake gcc g++ qt4-qmake libqt4-dev \
libusb-dev libftdi-dev \
python3-defusedxml python3-vcstool \
ros-kinetic-octomap-msgs        \
ros-kinetic-joy                 \
ros-kinetic-geodesy             \
ros-kinetic-octomap-ros         \
ros-kinetic-control-toolbox     \
ros-kinetic-pluginlib	       \
ros-kinetic-trajectory-msgs     \
ros-kinetic-control-msgs	       \
ros-kinetic-std-srvs 	       \
ros-kinetic-nodelet	       \
ros-kinetic-urdf		       \
ros-kinetic-rviz		       \
ros-kinetic-kdl-conversions     \
ros-kinetic-eigen-conversions   \
ros-kinetic-tf2-sensor-msgs     \
ros-kinetic-pcl-ros \
ros-kinetic-navigation
```
```
sudo apt-get install ros-kinetic-ar-track-alvar-msgs
sudo apt-get install ros-kinetic-sophus
sudo apt-get install git
```

**CATKIN IGNORE the 3 mentioned packages**
- spacenav_node not compiling. CATKIN_IGNOREd.
- wiimote not compiling. CATKIN_IGNOREd.
- kobuki_qtestsuite not compiling. CATKIN_IGNOREd.

**Add 3 more packages to the file gazebo.repos locates in /home/zshaikh/projects/gym-gazebo/gym_gazebo/envs . thes packages were related to open cv for ros**
 ```
 geometry:
    type: git
    url: https://github.com/ros/geometry
    version: indigo-devel
  geometry2:
    type: git
    url: https://github.com/ros/geometry2
    version: indigo-devel
  vision_opencv:
    type: git
    url: https://github.com/ros-perception/vision_opencv.git
    version: kinetic
```

```
cd gym_gazebo/envs/installation
bash setup_kinetic.bash	
bash turtlebot_setup.bash
```

**Import error : No module 'em'---Refer  [this](https://answers.ros.org/question/257331/python-module-empy-missing-tutorials)**
`pip install empy`

**During compilation you will require to solve some python3 errors related print statements of python2.You can do this with the below commands**

```
2to3 -w /home/zshaikh/projects/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/control_toolbox/cfg/Parameters.cfg

2to3 -w /home/zshaikh/projects/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/spawn_model

2to3 -w /home/zshaikh/projects/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros/gazebo_interface.py
```
**parameter error while running turtlebot example.Refer https://github.com/erlerobot/gym-gazebo/issues/58**
`gedit /home/zshaikh/projects/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/turtlebot_simulator/turtlebot_gazebo/launch/includes/kobuki.launch.xml`
Remove '--inorder' from line 6

yangningbocn, 2018-01-25:
I used python2 with ubuntu14.04 before.But there are some BUGs,That i can't slove. Can i use python2 with ubuntu16.04 ? Using pthon3, i should change so many things. Can it work?

zishanahmed08, 2018-01-25:
Yes.I have tested it on python2 Ubuntu 16.04 also.the installation is much easier than and a lot of the steps mentioned here are not required.Just stick to the the authors notes in install.md

yangningbocn, 2018-01-25:
Following authors notes, when we run "python circuit2_turtlebot_lidar_qlearn.py",Are we using python2.7 or python3.5,i think it is python2.7 ?Does ROS have python-dist for python3.5?I met the error can't import rospy with python3.5 before.

zishanahmed08, 2018-01-25:
**Following authors notes, when we run "python circuit2_turtlebot_lidar_qlearn.py",Are we using python2.7 or python3.5,i think it is python2.7**
The author runs it on python3.But you can run it for python2. It depends on the compilation of catkin-ws at the setup_kinetic.bash. Was it doen on virtual environment of python2 or python3?

**Does ROS have python-dist for python3.5?I met the error can't import rospy with python3.5 before.**
No.You will have to compile it from source.That is exactly what you are doing in your catkin_ws.However rospy and catkin pkgs exist for both py2 and py3.but they cannot co-exist.

In simple words, if you want the experiment for python2 - run setup_kinetic.bash in py2 virtual env and for python3 - run setup_kinetic.bash in py3 virtual env

yangningbocn, 2018-01-25:
Your answer is very clear,I understand by your explan,I decide to follow you and author.Try to setup it in py3. Thank you very much!

hlinh0411hd, 2018-01-30:
Hi I has a question about errors after updating Ubuntu. 
Before that my computer runs gym gazebo normally but after updating Ubuntu it can't work.
I also try to remove all ROS, gazebo, gym and install again but nothing works.
I've got that issue with 4 computer. Anyone has idea?

HailinRen, 2018-02-01:
[gym-gazebo.txt](https://github.com/erlerobot/gym-gazebo/files/1686576/gym-gazebo.txt)

ZhangWei1993, 2018-02-05:
Thank you very much for sharing! have you encountered the "ImportError: dynamic module does not define module export function (PyInit_cv_bridge_boost)" error when using cnn? I think cv_bridge fails to compile with Python 3, but I don't know how to solve it.

closed 2018-02-18 by vmayoral

KuangHaofei, 2018-06-06:
Hi,
I fellow your installation notes, but i encounter below problem while i execute  `bash setup_kinetic.bash
`:
/usr/include/gazebo-8/gazebo/msgs/param.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is
  ^
/usr/include/gazebo-8/gazebo/msgs/param.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
 #error incompatible with your Protocol Buffer headers.  Please
  ^
/usr/include/gazebo-8/gazebo/msgs/param.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.

How do i solve it? Thank you very much!

ElizabethDuo, 2018-08-10:
I have ROS and Gazebo in original python2.7.
Can I reinstall ROS and Gazebo in the conda environment?

jestifizzle, 2018-08-18:
I'm on Ubuntu 16.04 using python2.7, followed all the installation steps with dependencies, but my catkin_ws/src folder after installation is empty except for the CMakeLists.txt. Due to that, all the setup.bash files don't work that use the catkin_ws directory since there are no folders in it. Why is the catkin_ws empty?

srikanth-kilaru, 2018-10-18:
Hi
I am getting the following error even after following all the steps. Can you please help
bash: /home/srikanth/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel/setup.bash: No such file or directory

HailinRen, 2018-11-01:
@srikanth-kilaru  You need build the catkin_ws at least one time to generate the "devel" folder with "setup.bash".