erlerobot/gym-gazebo #91: Asynchronous Actor-Critic Agents (A3C)

hlinh0411hd, 2018-01-30:
Hi. I'm trying to implement A3C algorithm for turtlebot. According to algorithm, there are some workers which has own copy of environment but I think it can't run more than one gazebo or ROS in a computer. So I tried to using an environment and each time a worker run action, robot would change to position that worker has saved. Do you have other idea or some examples for that algorithm?
Thank you

vmayoral, 2018-01-31:
Hi @hlinh0411hd, A3C implementations available are for discrete action spaces so we're not really putting al lot of time into it. Let us know if you can provide a continuous state/action space implementation that's proven to work and we'd be happy to run a few benchmarks for you.

vmayoral, 2018-02-18:
Closing for now since there's been no further answer, nor interest apparently.

closed 2018-02-18 by vmayoral