erlerobot/gym-gazebo #92: Recently I successfully install the gym gazebo,but when I run a simple q learning example , I encountered the following error. Is there any solution to fix this problem,please.

undersunshine, 2018-02-01:
ImportError:
Importing the multiarray numpy extension module failed.  Most
likely you are trying to import a failed build of numpy.
If you're working with a numpy git repo, try `git clean -xdf` (removes all
files not under version control).  Otherwise reinstall numpy.

Original error was: cannot import name multiarray

[spawn_turtlebot_model-2] process has died [pid 5610, exit code 1, cmd /home/yaozijie/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/spawn_model -unpause -urdf -param robot_description -model mobile_base __name:=spawn_turtlebot_model __log:=/home/yaozijie/.ros/log/a3825982-072b-11e8-a4c0-00e0704e03ba/spawn_turtlebot_model-2.log].
log file: /home/yaozijie/.ros/log/a3825982-072b-11e8-a4c0-00e0704e03ba/spawn_turtlebot_model-2*.log
Segmentation fault (core dumped)
[gazebo-1] process has died [pid 5606, exit code 139, cmd /home/yaozijie/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -e ode home/user/gym-gazebo/gym_gazebo/envs/assets/worlds/circuit2.world __name:=gazebo __log:=/home/yaozijie/.ros/log/a3825982-072b-11e8-a4c0-00e0704e03ba/gazebo-1.log].
log file: /home/yaozijie/.ros/log/a3825982-072b-11e8-a4c0-00e0704e03ba/gazebo-1*.log

vmayoral, 2018-02-18:
You need to provide more details if we're to reproduce/help/support you.

zhuzd12, 2018-05-08:
This has happened to me before, and I just reinstalled numpy to get it fixed.

nzlz, 2019-01-15:
Seems like its not a gym-gazebo related error. Solved and closing.

closed 2019-01-15 by nzlz