erlerobot/gym-gazebo #93: ImportError: dynamic module does not define module export function (PyInit_cv_bridge_boost)

ZhangWei1993, 2018-02-05:
I defined my own environment and run it using cnn. however, I encounter some problems when I want to import Cv_Bridge. Does anyone encounter the same problem, and how to solve it? Thanks
`Traceback (most recent call last):
  File "dueling_dqn.py", line 60, in <module>
    state,done = env.reset()
  File "/home/zhw/.local/lib/python3.5/site-packages/gym/core.py", line 104, in reset
    return self._reset()
  File "/home/zhw/gym-gazebo/gym_gazebo/envs/turtlebot/gazebo_room1_turtlebot_lidar.py", line 133, in _reset
    cv_image = CvBridge().imgmsg_to_cv2(image_data, "bgr8")
  File "/home/zhw/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/vision_opencv/cv_bridge/python/cv_bridge/core.py", line 163, in imgmsg_to_cv2
    dtype, n_channels = self.encoding_to_dtype_with_channels(img_msg.encoding)
  File "/home/zhw/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/vision_opencv/cv_bridge/python/cv_bridge/core.py", line 99, in encoding_to_dtype_with_channels
    return self.cvtype2_to_dtype_with_channels(self.encoding_to_cvtype2(encoding))
  File "/home/zhw/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/vision_opencv/cv_bridge/python/cv_bridge/core.py", line 91, in encoding_to_cvtype2
    from cv_bridge.boost.cv_bridge_boost import getCvType
ImportError: dynamic module does not define module export function (PyInit_cv_bridge_boost)
[spawn_turtlebot_model-2] process has finished cleanly
log file: /home/zhw/.ros/log/1a1daff8-0a45-11e8-b0a1-f85971482dd5/spawn_turtlebot_model-2*.log`

zishanahmed08, 2018-02-05:
Step 1.  rename cv2.so to another name(cv2-renamed.so) at /opt/ros/kinetic/lib/python2.7/dist-packages/
Step 2. pip install opencv-python in your virtual env python3.5
Step 3. Add 3 more packages to the file gazebo.repos located in /home/zshaikh/projects/gym-gazebo/gym_gazebo/envs . these packages were related to open cv for ros
 ```
 geometry:
    type: git
    url: https://github.com/ros/geometry
    version: indigo-devel
  geometry2:
    type: git
    url: https://github.com/ros/geometry2
    version: indigo-devel
  vision_opencv:
    type: git
    url: https://github.com/ros-perception/vision_opencv.git
    version: kinetic
```
Refer this if you get further errors https://github.com/erlerobot/gym-gazebo/issues/90

ZhangWei1993, 2018-02-05:
Thank you! Actually, I have tried step1 and 3 already, but it didn't work. I think I need to install opencv in virtual env python3.5 because I didn't use conda install before.

vmayoral, 2018-02-18:
@ZhangWei1993, I'm closing this issue. Feel free to re-open it or comment if you have further issues.

closed 2018-02-18 by vmayoral