erlerobot/gym-gazebo #94: Import importlib error when executing 'python3 circuit2_turtlebot_lidar_qlearn.py' from virtualevn.

max99diving, 2018-02-09:
I managed to build the entire project after several corrections. When I now run the first example, circuit2_turtlebot_lidar_qlearn.py, from within virtualenv, I get a run time error. The error is reported below. The first error seems to be about 'import importlib', but if I try to just execute python3 import importlib there are no issues!!!
I would be forever grateful if someone could tell me what's going on here.

(Gym-Gazebo) username@ukmrdd-55256:~/Gym-Gazebo/gym-gazebo/examples/turtlebot$ python3 circuit2_turtlebot_lidar_qlearn.py
... logging to /home/username/.ros/log/c81a29c2-0d9e-11e8-a800-d481d7ab8bf1/roslaunch-ukmrdd-55256-3562.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ukmrdd-55256:39852/
ros_comm version 1.12.12

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12

NODES

auto-starting new master
process[master]: started with pid [3591]
ROS_MASTER_URI=http://ukmrdd-55256:11311/

setting /run_id to c81a29c2-0d9e-11e8-a800-d481d7ab8bf1
process[rosout-1]: started with pid [3615]
started core service [/rosout]
Roscore launched!
Gazebo launched!
... logging to /home/username/.ros/log/c81a29c2-0d9e-11e8-a800-d481d7ab8bf1/roslaunch-ukmrdd-55256-3640.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/home/username/Gym-Gazebo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/xacro/xacro.py", line 60, in <module>
    import xacro
  File "/home/username/Gym-Gazebo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel/lib/python3/dist-packages/xacro/__init__.py", line 8, in <module>
    from pkgutil import extend_path
  File "/usr/lib/python3.5/pkgutil.py", line 4, in <module>
    import importlib
ImportError: No module named 'importlib'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 53, in apport_excepthook
    if not enabled():
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 24, in enabled
    import re
ImportError: No module named 're'

Original exception was:
Traceback (most recent call last):
  File "/home/username/Gym-Gazebo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/xacro/xacro.py", line 60, in <module>
    import xacro
  File "/home/username/Gym-Gazebo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel/lib/python3/dist-packages/xacro/__init__.py", line 8, in <module>
    from pkgutil import extend_path
  File "/usr/lib/python3.5/pkgutil.py", line 4, in <module>
    import importlib
ImportError: No module named 'importlib'
while processing /home/username/Gym-Gazebo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/turtlebot_simulator/turtlebot_gazebo/launch/includes/kobuki.launch.xml:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/home/mgiammatteo/Gym-Gazebo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/xacro/xacro.py '/home/mgiammatteo/Gym-Gazebo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/turtlebot/turtlebot_description/robots/kobuki hexagons_asus_xtion_pro.urdf.xacro' ] returned with code [1]. 

Param xml is <param command="$(arg urdf_file)" name="robot_description"/>
The traceback for the exception was written to the log file

vmayoral, 2018-02-18:
You're missing this package:

>import re
>ImportError: No module named 're'

vmayoral, 2018-02-18:
Review your output, fix the dependencies and open a new issue if you're still stucked.

closed 2018-02-18 by vmayoral