erlerobot/gym-gazebo #96: Dockerfile updates

AustinDeric, 2018-02-12:
It may simplify erlebot's Dockerfile to use the ros docker images.  I can help this effort if there is interest.

I was going to make a dockerfile to run the SCARA demos. The documentation says the docker method is deprcated, I am curious to know why.  Is it due to a different strategic direction or just not enough man-power for support?

Also, this repo is great work, thanks!

vmayoral, 2018-02-13:
Hello @AustinDeric,

Thanks for https://github.com/erlerobot/gym-gazebo/pull/95. Unfortunately, the SCARA source files are not open source (at least not fully) since we're still working on it. That said, there's already a decent amount of infrastructure laid and opened so that you can extent what exists and create your own SCARA robot. Alternatively, for a quick start, refer to https://github.com/erlerobot/gym-gazebo#community-maintained-environments for the community-maintained environments.

> The documentation says the docker method is deprcated, I am curious to know why. Is it due to a different strategic direction or just not enough man-power for support?

Certainly the latter. Lately, we've been trying to allocate time to simplify things for potential users. I can't commit to anything but we have a few interesting internal ideas that can certainly improve significantly the whole process of using gym-gazebo.

> It may simplify erlebot's Dockerfile to use the ros docker images. I can help this effort if there is interest.

That'll be very welcome.

AustinDeric, 2018-02-13:
Awesome, i will help out with the docker stuff.  It may be useful to replace the SCARA robot with a universal robot:  https://github.com/ros-industrial/universal_robot

there is a lot of support for this platform in the research community.

As an aside, have ya'll had the following issue:
[image]

I'll figure it out, but i am always open to saving time:

zishanahmed08, 2018-02-15:
@AustinDeric - Sorry for hijacking your issue. Could please give your comments on 
 a new docker related gym-gazebo query which i just created here https://github.com/erlerobot/gym-gazebo/issues/98

mkhansen-intel, 2018-04-06:
@AustinDeric - Have you made any progress towards a new dockerfile? I'm looking into the Travis CI and trying to get that working again, but when I'm trying to build a new dockerfile it's failing. I've never created a dockerfile before so I'm a noob to that. If you can create a working dockerfile, I think I can get the CI working. Let me know.

AustinDeric, 2018-05-09:
Hey guys, i dropped the ball on this one, i wasn't watching.  I will update ya'll shortly.

mkhansen-intel, 2018-05-11:
@AustinDeric - do you have an ETA for a new dockerfile? As I said above, if so, I'd like to try and get the CI working using the new file.

nzlz, 2019-01-15:
Continued at https://github.com/erlerobot/gym-gazebo/pull/165. Closing.

closed 2019-01-15 by nzlz