erlerobot/gym-gazebo #98: Multiple gym-gazebo (ROS & Gazebo )experiments with Docker

zishanahmed08, 2018-02-15:
One of the drawbacks of Gym-gazebo is that we cannot run more than one experiment at a time since  we can only have one ROS-master running at a time. I tried a solution run multiple ros/gazebo instances [here](https://answers.ros.org/question/193062/how-to-run-multiple-independent-gazebo-instances-on-the-same-machine).But could not get it working for gym-gazebo.

My next approach would be using dockers to run multiple simultaneous experiment.Before investing time & effort in creating(modifying) the docker, I would like to know if it is theoretically possible to multiple dockers at a time and therefore running multiple ROS/Gazebo instances:has anyone tried this?
 
PS- I have never used Docker before.Please excuse if this question sounds stupid .

AustinDeric, 2018-02-18:
yes, you can do this.  I would use the python docker api to create the docker containers in bulk https://github.com/docker/docker-py.  I am not sure this should be an issue with this repo.

zishanahmed08, 2018-02-18:
Thanks @AustinDeric  ..FYI @pmwenzel

vmayoral, 2018-02-18:
As pointed out, it is certainly possible. Check https://github.com/erlerobot/gym-gazebo/blob/master/Dockerfile. It provides a good starting point.

closed 2018-02-18 by vmayoral