erlerobot/gym-gazebo #99: Multiple gym-gazebo (ROS & Gazebo )experiments by modifying ROS_MASTER_URI & GAZEBO_MASTER_URI

zishanahmed08, 2018-02-18:
I tried a solution run multiple ros/gazebo instances [here](https://answers.ros.org/question/193062/how-to-run-multiple-independent-gazebo-instances-on-the-same-machine/) . .But could not get it working for gym-gazebo.
Could you please let me know if you have tried this method for running multiple experiments.
I guess the below changes would be reuired
1)Exporting ROS_MASTER_URI & GAZEBO_MASTER_URI in entrypoint.sh
https://github.com/erlerobot/gym-gazebo/blob/c8783dee491168f88f2469a3c3afb47dc552a093/entrypoint.sh#L9

2)Modifications in gazebo_env
https://github.com/erlerobot/gym-gazebo/blob/c8783dee491168f88f2469a3c3afb47dc552a093/gym_gazebo/envs/gazebo_env.py#L20

3) Modify GAZEBO_MASTER_URI in some gazebo scripts in gazebo_ros located in  gym_gazebo/envs/installation/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/gazebo for example

Any comments/help on this would be appreciated

hlinh0411hd, 2018-03-06:
Each terminal, I ran 
`export ROS_MASTER_URI=http://localhost:11315 ROS_PORT_SIM=11315
export GAZEBO_MASTER_URI=http://localhost:11355
`

with different number port and it worked.

zishanahmed08, 2018-03-07:
@pmwenzel

zishanahmed08, 2018-03-26:
@hlinh0411hd - It works initally but after a couplke of minutes i get timeout errors related to lasescan.Any ideas how to solve this?

> rospy.exceptions.ROSException: timeout exceeded while waiting for message on topic /scan

nzlz, 2019-01-15:
By default the `gazebo_env.py` now creates the network segmentation you are talking about, setting different ROS_MASTER_URI and GAZEBO_MASTER_URI to each execution. This should fix this issue, closing.

closed 2019-01-15 by nzlz