erlerobot/gym-gazebo #101: Can not build workspace with packages of unknown build_type

salfityj, 2018-02-22:
Ubuntu 16 and ROS Kinetic
Using an anaconda environment `salfity_env` with python3.5.

After cloning the gym-gazebo repo, i'm trying `catkin_make` , within the `setup_kinetic.bash`

Error message: ` This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.`

Confused where all these packages came from, because when I built this repo with ROS Indigo on Ubuntu 14.04, they don't seem to be there.

Haven't seen this error with ROS and catkin before, any suggestions?
Yes, I did try `catkin_make_isolated`

```
(salfity_env) salfity@gravitational:~/gym-gazebo/gym_gazebo/envs/installation/catkin_ws$ catkin_make -j6
Base path: /home/salfity/gym-gazebo/gym_gazebo/envs/installation/catkin_ws
Source space: /home/salfity/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src
Build space: /home/salfity/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build
Devel space: /home/salfity/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel
Install space: /home/salfity/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/install
WARNING: Metapackage "ecl_lite" must buildtool_depend on catkin.
WARNING: Metapackage "ecl_lite" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
WARNING: Metapackage "ecl_tools" must buildtool_depend on catkin.
WARNING: Metapackage "ecl_tools" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
####
#### Running command: "cmake /home/salfity/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/salfity/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/salfity/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/install -G Unix Makefiles" in "/home/salfity/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/salfity/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/salfity/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/salfity/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /home/salfity/anaconda3/envs/salfity_env/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/salfity/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.7.8
-- BUILD_SHARED_LIBS is on
WARNING: Metapackage "ecl_tools" must buildtool_depend on catkin.
WARNING: Metapackage "ecl_tools" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
WARNING: Metapackage "ecl_lite" must buildtool_depend on catkin.
WARNING: Metapackage "ecl_lite" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
WARNING: Metapackage "ecl_tools" must buildtool_depend on catkin.
WARNING: Metapackage "ecl_tools" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
WARNING: Metapackage "ecl_lite" must buildtool_depend on catkin.
WARNING: Metapackage "ecl_lite" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 150 packages in topological order:
-- ~~  - catkin_simple
-- ~~  - create_description
-- ~~  - create_driver
-- ~~  - driver_common (metapackage)
-- ~~  - ecl_core (metapackage)
-- ~~  - ecl_license (plain cmake)
-- ~~  - ecl_build (plain cmake)
-- ~~  - ecl_command_line
-- ~~  - ecl_config (plain cmake)
-- ~~  - ecl_console (plain cmake)
-- ~~  - ecl_converters_lite (plain cmake)
-- ~~  - ecl_eigen
-- ~~  - ecl_errors (plain cmake)
-- ~~  - ecl_exceptions
-- ~~  - ecl_filesystem
-- ~~  - ecl_io (plain cmake)
-- ~~  - ecl_lite (metapackage)
WARNING: The CMakeLists.txt of the metapackage 'ecl_lite' contains non standard content. Use the content of the following file instead: /home/salfity/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/catkin_generated/metapackages/ecl_lite/CMakeLists.txt
-- ~~  - ecl_mpl
-- ~~  - ecl_navigation (metapackage)
-- ~~  - ecl_sigslots_lite (plain cmake)
-- ~~  - ecl_time_lite (plain cmake)
-- ~~  - ecl_time
-- ~~  - ecl_ipc
-- ~~  - ecl_tools (metapackage)
WARNING: The CMakeLists.txt of the metapackage 'ecl_tools' contains non standard content. Use the content of the following file instead: /home/salfity/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/catkin_generated/metapackages/ecl_tools/CMakeLists.txt
-- ~~  - ecl_type_traits
-- ~~  - ecl_concepts
-- ~~  - ecl_converters
-- ~~  - ecl_formatters
-- ~~  - ecl_math
-- ~~  - ecl_utilities
-- ~~  - ecl_containers
-- ~~  - ecl_threads
-- ~~  - ecl_devices
-- ~~  - ecl_sigslots
-- ~~  - ecl_streams
-- ~~  - gazebo_ros_pkgs (metapackage)
-- ~~  - hector_gazebo (metapackage)
-- ~~  - hector_gazebo_worlds
-- ~~  - hector_sensors_gazebo
-- ~~  - image_common (metapackage)
-- ~~  - joystick_drivers (metapackage)
-- ~~  - kobuki (metapackage)
-- ~~  - kobuki_capabilities
-- ~~  - kobuki_core (metapackage)
-- ~~  - kobuki_desktop (metapackage)
-- ~~  - kobuki_ftdi
-- ~~  - kobuki_gazebo
-- ~~  - kobuki_rapps
-- ~~  - kobuki_rviz_launchers
-- ~~  - navigation (metapackage)
-- ~~  - perception_pcl (metapackage)
-- ~~  - ros_control (metapackage)
-- ~~  - roslint
-- ~~  - ecl_linear_algebra
-- ~~  - ecl_geometry
-- ~~  - ecl_core_apps
-- ~~  - ecl_mobile_robot
-- ~~  - ecl_statistics
-- ~~  - kobuki_dock_drive
-- ~~  - kobuki_driver
-- ~~  - controller_manager_msgs
-- ~~  - kobuki_msgs
-- ~~  - move_base_msgs
-- ~~  - gazebo_msgs
-- ~~  - turtlebot (metapackage)
-- ~~  - turtlebot_bringup
-- ~~  - turtlebot_capabilities
-- ~~  - turtlebot_create (metapackage)
-- ~~  - turtlebot_gazebo
-- ~~  - turtlebot_simulator (metapackage)
-- ~~  - turtlebot_stage
-- ~~  - turtlebot_stdr
-- ~~  - camera_calibration_parsers
-- ~~  - hardware_interface
-- ~~  - controller_interface
-- ~~  - kobuki_keyop
-- ~~  - kobuki_bumper2pc
-- ~~  - ps3joy
-- ~~  - realtime_tools
-- ~~  - controller_manager
-- ~~  - joint_limits_interface
-- ~~  - image_transport
-- ~~  - camera_info_manager
-- ~~  - polled_camera
-- ~~  - rqt_controller_manager
-- ~~  - spacenav_node
-- ~~  - timestamp_tools
-- ~~  - control_toolbox
-- ~~  - controller_manager_tests
-- ~~  - driver_base
-- ~~  - joy
-- ~~  - kobuki_auto_docking
-- ~~  - create_node
-- ~~  - fake_localization
-- ~~  - gazebo_ros
-- ~~  - hector_gazebo_plugins
-- ~~  - kobuki_dashboard
-- ~~  - map_server
-- ~~  - amcl
-- ~~  - pcl_ros
-- ~~  - robot_pose_ekf
-- ~~  - pointcloud_to_laserscan
-- ~~  - transmission_interface
-- ~~  - turtlebot_teleop
-- ~~  - gazebo_plugins
-- ~~  - gazebo_ros_control
-- ~~  - hector_gazebo_thermal_camera
-- ~~  - kobuki_gazebo_plugins
-- ~~  - voxel_grid
-- ~~  - costmap_2d
-- ~~  - nav_core
-- ~~  - base_local_planner
-- ~~  - carrot_planner
-- ~~  - clear_costmap_recovery
-- ~~  - dwa_local_planner
-- ~~  - move_slow_and_clear
-- ~~  - navfn
-- ~~  - global_planner
-- ~~  - rotate_recovery
-- ~~  - move_base
-- ~~  - wiimote
-- ~~  - xacro
-- ~~  - kobuki_description
-- ~~  - turtlebot_description
-- ~~  - yocs_ar_pair_approach
-- ~~  - yocs_cmd_vel_mux
-- ~~  - yocs_controllers
-- ~~  - kobuki_controller_tutorial
-- ~~  - kobuki_random_walker
-- ~~  - kobuki_safety_controller
-- ~~  - yocs_keyop
-- ~~  - yocs_localization_manager
-- ~~  - yocs_math_toolkit
-- ~~  - yocs_ar_marker_tracking
-- ~~  - yocs_diff_drive_pose_controller
-- ~~  - yocs_msgs
-- ~~  - yocs_ar_pair_tracking
-- ~~  - yocs_joyop
-- ~~  - yocs_navi_toolkit
-- ~~  - yocs_navigator
-- ~~  - yocs_rapps
-- ~~  - yocs_safety_controller
-- ~~  - yocs_velocity_smoother
-- ~~  - kobuki_node
-- ~~  - kobuki_testsuite
-- ~~  - kobuki_qtestsuite
-- ~~  - yocs_virtual_sensor
-- ~~  - yocs_waypoint_provider
-- ~~  - yocs_waypoints_navi
-- ~~  - yujin_ocs (metapackage)
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:63 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/salfity/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/salfity/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
'''

max99diving, 2018-02-23:
I also have this problem and it seems a recent one as previously only experienced issues at run-time.

xueeinstein, 2018-02-24:
@salfityj @max99diving  I also have this problem. Fortunately, I found that @stonier's recent commits in projects [ecl_tools](https://github.com/stonier/ecl_tools) and [ecl_lite](https://github.com/stonier/ecl_lite) broke the code on master branch. To fix this, I checkout to the recent tag commit. Hope this could help you as well.

max99diving, 2018-02-24:
Sorry not very familiar with GitHub, not sure what you mean when you say that you checked out to the recent tag commit. How do I identify the 'recent tag commit'?

xueeinstein, 2018-02-24:
Check the release page, like [this](https://github.com/stonier/ecl_tools/releases). Each tag commit records a stable minor version.

stonier, 2018-02-24:
I'm slowly migrating packages over to ament / ROS2 so the devel branches are no longer catkin capable. Make sure you use the release branches or tags instead (e.g. [release/0.61-indigo-kinetic](https://github.com/stonier/ecl_tools/tree/release/0.61-indigo-kinetic)). Good practice to use those anyway since changes on a devel or master branch will always bite you.

salfityj, 2018-04-10:
Yup, thanks Stonier, that was the fix, thanks!
Need to change the tags in `gazebo.repos`

closed 2018-04-10 by salfityj