erlerobot/gym-gazebo #105: Actions vs. resultant state

rdockterjr, 2018-02-26:
This isn't so much a code bug but perhaps an algorithmic issue and please let me know if I am thinking about this wrong.

Specifically I am looking at the gazebo_cartpole implementation but this may be the case among all the environments. My question is this:

When I call step() with a given action, I would expect that the simulation should apply the action then wait to see what the resultant new state is. i.e. the state that was caused by the action.

However around line 66 in envs/gazebo_cartpole/gazebo_cartpole_v0.py it looks like the step function just checks if self.data is None. Doesn't this mean that the current data could be slightly stale if it was updated by the callback() just before I applied the action?

I would think the following code would more explicitly force the next state to be the result of the current action:

self.data = None
while self.data is None:
try:
   self.data = rospy.wait_for_message('/joint_states', JointState, timeout=5)
except:
   pass

vmayoral, 2018-02-26:
@ahcorde might want to take this one.

nzlz, 2019-01-15:
Thanks for the feedback @rdockterjr ,  you are right, the code related to `/cart_pole/joint_states` is not correct, the callback function should be removed. We will take this into account if we decide to migrate the Cartpole to the coming ROS2 version of gym-gazebo.

closed 2019-01-15 by nzlz