erlerobot/gym-gazebo #106: ImportError: No module named baselines.agent.scara_arm.tree_urdf

mkhansen-intel, 2018-03-07:
I'm trying to run the 4dof articulated arm simulation, and this is the error I'm getting.
```
~/gym-gazebo$ python gym_gazebo/envs/articulated_arm/gazebo_modular_articulated_arm_4dof_v1.py 
Traceback (most recent call last):
  File "gym_gazebo/envs/articulated_arm/gazebo_modular_articulated_arm_4dof_v1.py", line 18, in <module>
    from baselines.agent.scara_arm.tree_urdf import treeFromFile # For KDL Jacobians
ImportError: No module named baselines.agent.scara_arm.tree_urdf
```
These are the files I see when I search for it locally:
```
~/gym-gazebo$ find . -name *urdf* | grep scara
./gym_gazebo/envs/assets/urdf/modular_scara/scara_e1_model_4_and_3joints.urdf.xacro
./gym_gazebo/envs/assets/urdf/modular_scara/scara_e1_4joints.urdf
./gym_gazebo/envs/assets/urdf/modular_scara/scara_e1_model_4joints.urdf.xacro
./gym_gazebo/envs/assets/urdf/modular_scara/scara_e1_3joints.urdf
./gym_gazebo/envs/assets/urdf/modular_scara/scara_e1_model_3joints.urdf.xacro
./gym_gazebo/envs/assets/urdf/modular_scara/scara_e1_model_4_and_3joints.urdf
```
I've tried searching the repo and the web for the urdf file, but no luck. Can someone point me to it?

vmayoral, 2018-03-22:
We're still working to release the modular_scara files. It isn't ready just yet.

We'll announce it and move this environment to the community-supported ones once available.

closed 2018-03-22 by vmayoral

MrDadaGuy, 2019-02-27:
Hi all, any update?  Thanks!

nzlz, 2019-02-27:
Release of gym-gazebo2 coming **really soon**. Which will be focused on [MARA](https://acutronicrobotics.com/docs/products/mara). Stay tuned!!