erlerobot/gym-gazebo #110: CMake Error:Workspace contains non-catkin packages in it

Bobobs, 2018-03-08:
I've experience the follwing cmake error when launching the setup_kinetic.bash file
```
bob@BOB:~/gym-gazebo/gym_gazebo/envs/installation$ bash setup_kinetic.bash 
Creating symlink "/home/bob/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/CMakeLists.txt" pointing to "/opt/ros/kinetic/share/catkin/cmake/toplevel.cmake"
..........................
=== ./catkin_simple (git) ===
Cloning into '.'...
Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== ./control_toolbox (git) ===
Cloning into '.'...
Switched to a new branch 'indigo-devel'
Branch indigo-devel set up to track remote branch indigo-devel from origin.
=== ./driver_base (git) ===
Cloning into '.'...
Already on 'indigo-devel'
Your branch is up-to-date with 'origin/indigo-devel'.
=== ./ecl_core (git) ===
Cloning into '.'...
Already on 'devel'
Your branch is up-to-date with 'origin/devel'.
=== ./ecl_lite (git) ===
Cloning into '.'...
Already on 'devel'
Your branch is up-to-date with 'origin/devel'.
=== ./ecl_navigation (git) ===
Cloning into '.'...
Already on 'devel'
Your branch is up-to-date with 'origin/devel'.
=== ./ecl_tools (git) ===
Cloning into '.'...
Already on 'devel'
Your branch is up-to-date with 'origin/devel'.
=== ./gazebo_ros_pkgs (git) ===
Cloning into '.'...
Switched to a new branch 'indigo-devel'
Branch indigo-devel set up to track remote branch indigo-devel from origin.
=== ./hector_gazebo (git) ===
Cloning into '.'...
Switched to a new branch 'indigo-devel'
Branch indigo-devel set up to track remote branch indigo-devel from origin.
=== ./image_common (git) ===
Cloning into '.'...
Already on 'hydro-devel'
Your branch is up-to-date with 'origin/hydro-devel'.
=== ./joystick_drivers (git) ===
Cloning into '.'...
Already on 'indigo-devel'
Your branch is up-to-date with 'origin/indigo-devel'.
=== ./kobuki (git) ===
Cloning into '.'...
Switched to a new branch 'indigo'
Branch indigo set up to track remote branch indigo from origin.
=== ./kobuki_core (git) ===
Cloning into '.'...
Switched to a new branch 'indigo'
Branch indigo set up to track remote branch indigo from origin.
=== ./kobuki_desktop (git) ===
Cloning into '.'...
Switched to a new branch 'indigo'
Branch indigo set up to track remote branch indigo from origin.
=== ./kobuki_msgs (git) ===
Cloning into '.'...
Switched to a new branch 'indigo'
Branch indigo set up to track remote branch indigo from origin.
=== ./navigation (git) ===
Cloning into '.'...
Switched to a new branch 'indigo-devel'
Branch indigo-devel set up to track remote branch indigo-devel from origin.
=== ./pcl_ros (git) ===
[... 246 lines cut ...]
-- ~~  - yocs_navigator
-- ~~  - yocs_rapps
-- ~~  - yocs_safety_controller
-- ~~  - yocs_velocity_smoother
-- ~~  - kobuki_node
-- ~~  - kobuki_testsuite
-- ~~  - kobuki_qtestsuite
-- ~~  - yocs_virtual_sensor
-- ~~  - yocs_waypoint_provider
-- ~~  - yocs_waypoints_navi
-- ~~  - yujin_ocs (metapackage)
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:63 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/bob/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/bob/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
## ROS workspace compiled ##
```

hlinh0411hd, 2018-03-09:
you change file gazebo.repos here:
`  ecl_lite:
    type: git
    url: https://github.com/stonier/ecl_lite
    version: release/0.61-indigo-kinetic
  ecl_navigation:
    type: git
    url: https://github.com/stonier/ecl_navigation
    version: devel
  ecl_tools:
    type: git
    url: https://github.com/stonier/ecl_tools
    version: release/0.61-indigo-kinetic`
then rerun setup.
Hope it helps you

Bobobs, 2018-03-09:
@hlinh0411hd That did the trick. Thank you

closed 2018-03-09 by Bobobs

TrinhTUHH, 2018-04-26:
I got the same error and following @hlinh0411hd suggestion did not help. Any idea?

TrinhTUHH, 2018-04-26:
I solved it. Need to change version of `ecl_core` to `release/0.61-indigo-kinetic` as well.

sandeepgogadi, 2018-06-29:
Add `sudo apt-get install ros-kinetic-ar-track-alvar ` to ROS kinetic dependencies.