erlerobot/gym-gazebo #112: Navigate to objects

hlinh0411hd, 2018-03-15:
Hi, I want to create a world with obstacles so turtlebot can find and navigate to the object I want. In that world, each episode, there is a different object has reward if turtlebot touch it and no reward or negative reward for all other objects. However, I still don't know how to make turtlebot know which object it touch. It means  when turtlebot hit an object, it'll return name of object it hit. Can you help me?
Thanks.

ahcorde, 2018-03-15:
You can detect collisions with the following code. Have a look to Contacts msg, I think you can know which objects are colliding.

```

/////////////////////////////////////////////////
// Function is called everytime a message is received.
void cb(ConstContactsPtr &_msg)
{
}

/////////////////////////////////////////////////
int main(int argc, char ** argv)
{
  // Load gazebo
  gazebo::client::setup(argc, argv);

  ros::init(argc, argv, "gazebo_contacts_node");
  ros::NodeHandle n;
	contacts_pub = n.advertise<gazebo_msgs::ContactState>("gazebo_contacts", 1);

  // Create our node for communication
  gazebo::transport::NodePtr node(new gazebo::transport::Node());
  node->Init();

  // Listen to Gazebo world_stats topic
  gazebo::transport::SubscriberPtr sub = node->Subscribe("/gazebo/default/physics/contacts", cb);

  // Busy wait loop...replace with your own code as needed.
  while (true)
    gazebo::common::Time::MSleep(10);

  // Make sure to shut everything down.
  gazebo::client::shutdown();
}
```

hlinh0411hd, 2018-03-15:
Thanks, I'll try

hlinh0411hd, 2018-03-16:
That's way we use topics from gazebo right? Is any solution to use ROS topic?
I found they said that there's no support python interface in gazebo, the pygazebo is too old.
How can I get Contacts msg in ROS by python?
Thanks again

nzlz, 2019-01-15:
In order to communicate with the gazebo c++ API you should do it via `gazebo_ros_pkgs` which provides an interface to communicate with ROS. Once gazebo_ros_pkgs creates the corresponding ros topis and services, you would be able to access that information using rospy (ros1) or rclpy (ros2).

If you require further advice, please open a **Advice request** issue to facilitate issue-tracking.

closed 2019-01-15 by nzlz