erlerobot/gym-gazebo #114: sudo: docker: command not found

mymultiverse, 2018-03-16:
I am trying to run  to connect the gzserver inside the container

    export GAZEBO_MASTER_IP=$(sudo docker inspect --format '{{ .NetworkSettings.IPAddress }}' "id of running container") 

But I am getting

    sudo: docker: command not found

LanderU, 2018-03-16:
Is docker available in your system? https://docs.docker.com/install/

ahcorde, 2018-03-16:
It's very complex to run gzclient inside a docker container. This line is for running gzclient on your host

mymultiverse, 2018-03-16:
@LanderU I have successfully installed docker. 
@ahcorde  When I try to run gzclient outside docker container in a new terminal happens nothing it automatically ends after a while.

ahcorde, 2018-03-16:
Did you change the id of your docker?

export GAZEBO_MASTER_IP=$(sudo docker inspect --format '{{ .NetworkSettings.IPAddress }}' **"id of running container"**) 

and then run
``` 
gzclient --verbose
```

mymultiverse, 2018-03-16:
Yes I changed the id but same error and for 
    
    gzclient --verbose

I am getting the following error both case inside the container and outside:-

```
Gazebo multi-robot simulator, version 7.3.1
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Err] [ConnectionManager.cc:119] Failed to connect to master in 30 seconds.
[Err] [gazebo_shared.cc:83] Unable to initialize transport.
[Err] [gazebo_client.cc:62] Unable to setup Gazebo

```
I am running environment before starting gazebo server.

mymultiverse, 2018-03-16:
I figured out some part but still don't know the reason. These commands need to be run outside the container. 
 ```
export GAZEBO_MASTER_IP=$(sudo docker inspect --format '{{ .NetworkSettings.IPAddress }}' "id of running container")
export GAZEBO_MASTER_URI=$GAZEBO_MASTER_IP:11345
gzclient
```

closed 2018-03-16 by mymultiverse