erlerobot/gym-gazebo #116: circuit2_turtlebot_lidar_dqn.py

ThorShockU, 2018-03-20:
@vmayoral @rkojcev @ahcorde @imuguruza @jlamperez 

I am always stuck at running the python script lidar_dqn.py where I got a sentence GazeboRoskobuki plugin ready to go! and then nothing pops out. However, everything is fine for running lidar_qlearn.py.
Any suggestions?
[image]

mkhansen-intel, 2018-03-21:
I saw this too, but when I scrolled up I saw other errors occurred before this point that actually stopped the python script. These messages are from the ROS processes that are launched in parallel. Scroll up and verify everything on the screen above this looks correct. I think you'll find an error occurred.

ahcorde, 2018-03-26:
Can you provide the trace of the terminal instead of the image? Probably the error is in upper lines.

Thanks

ThorShockU, 2018-04-17:
@mkhansen-intel Thx, I fixed it.

JaxDraxler, 2018-04-25:
@mkhansen-intel Once i had this issue, it resolved when i did the following:
bash turtlebot_setup.bash
bash turtlebot_nn_setup.bash
bash turtlebot_cnn_setup.bash

nzlz, 2019-01-15:
Only the latest sources would be applied (as they would overwrite the previous content of env variables), this is `turtlebot_cnn_setup.bash`, which contains the environment variables to work with the CNN algorithm. So be careful with that. I believe `bash turtlebot_setup.bash` would had been enough/correct in your case.

Since no further errors appeared, closing.

closed 2019-01-15 by nzlz