erlerobot/gym-gazebo #117: not  able to see copter in Gazebo simulation

mymultiverse, 2018-03-20:
I am loading and running the copter environment and example respectively with following commands, which run almost fine (with error message FCU: PreArm: RC not calibrated with UDP port log info.).

```sudo docker run -it erlerobotics/gym-gazebo
sudo pip install pandas
cd /usr/local/gym/gym-gazebo/gym_gazebo/envs/installation
xvfb-run -s "-screen 0 1400x900x24" bash
bash erlecopter_setup.bash
source catkin_ws/devel/setup.bash
cd /usr/local/gym/gym-gazebo/examples/scripts_erlecopter
python erlecopter_hover_qlearn.py
``` 

In the new terminal, I launch 

    gzclient

After launching gazebo I am getting this.
[image]

JaxDraxler, 2018-04-25:
@mymultiverse Hello !! Have you found a solution for this ?

mymultiverse, 2018-05-17:
@JaxDraxler not yet.

nzlz, 2019-01-15:
If any of you is still interested/facing this issue please open a **Bug report** to facilitate issue-tracking and we will try to help you.

closed 2019-01-15 by nzlz