erlerobot/gym-gazebo #119: [gazebo-1] process has died problem help me

yooseonghyuck, 2018-03-21:
-Run the environment with a sample agent:

>>cd gym_gazebo/examples/scripts_turtlebot
>>python3 circuit2_turtlebot_lidar_dqn.py

i have problem:

[gazebo-1] process has died [pid 2967, exit code 134, cmd /home/song/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -e ode home/user/gym-gazebo/gym_gazebo/envs/assets/worlds/circuit2.world __name:=gazebo __log:=/home/song/.ros/log/593daa28-2d16-11e8-8ee0-00c2c6dbfe1b/gazebo-1.log].
log file: /home/song/.ros/log/593daa28-2d16-11e8-8ee0-00c2c6dbfe1b/gazebo-1*.log

what shoud i do?

mkhansen-intel, 2018-03-21:
I've seen this when you have running processes (gazebo) on your system and you try to launch the script. I would try killing all the ros processes including gazebo, opening a fresh terminal and run again. 

From the README page, here is the command to kill all the ROS processes:
killall -9 rosout roslaunch rosmaster gzserver nodelet robot_state_publisher gzclient

yooseonghyuck, 2018-03-22:
@mkhansen-intel thanks for your comment.

but when i type that cammand, output is : 
gzserver: no process found
gzclient: no process found

it's strange. when i reboot ubuntu and launch python code, same error happens.
i haven't runned any ROS processes after rebooting ubuntu.

l don't know why ㅠㅠ

ahcorde, 2018-03-26:
With these lines: 

> gzserver: no process found
> gzclient: no process found

it seems that you don't have gazebo installed in your computer

hlinh0411hd, 2018-03-27:
@ahcorde it means there's no gzserver and gzclient are running.
I feel like some problems with python3 when running gym-gazebo. It conflicts with python in ROS.
I've seen that problem a lot of time.
So I usually reinstall Ubuntu and install gym gazebo but use python2 and install python2 packages instead

zacwellmer, 2018-05-10:
Any updates on how to sort this out? I'm also receiving this problem on the example scripts for turtlebot

nzlz, 2019-01-15:
Please stick to python2 or open a **Feature request** issue to facilitate issue-tracking. In addition, with the current network segmentation provided by gazebo_env.py, you should not face this issue now.

closed 2019-01-15 by nzlz