erlerobot/gym-gazebo #120: 'GazeboMazeTurtlebotLidarEnv' object has no attribute 'monitor'

mkhansen-intel, 2018-03-21:
There's a bug in maze_turtlebot_lidar_qlearn.py:

  File "maze_turtlebot_lidar_qlearn.py", line 29, in <module>
    env.monitor.start(outdir, force=True, seed=None)
AttributeError: 'GazeboMazeTurtlebotLidarEnv' object has no attribute 'monitor'

I know that if I comment out the env.monitor code, this will work. I see that has been done already in circuit2_turtlebot_lidar_qlearn.py. The real fix however according to the AI Gym Readme is this: 

2016-12-27: BACKWARDS INCOMPATIBILITY: The gym monitor is now a wrapper. Rather than starting monitoring as env.monitor.start(directory), envs are now wrapped as follows: env = wrappers.Monitor(env, directory). This change is on master and will be released with 0.7.0.

I can comment it out, but would rather submit a real fix here in a PR. However, since I see it was commented out in the other file, would like some advice on what to do about this.

mkhansen-intel, 2018-03-21:
Note that this appears in most of the Turtlebot examples, I assume all would need the same fix:
mkhansen@mkhansen-sky1:~/gym-gazebo/examples/turtlebot$ grep env *.py| grep monitor
circuit2c_turtlebot_camera_dqn.py:        env.monitor.start(outdir, force=True, seed=None)
circuit2c_turtlebot_camera_dqn.py:        env.monitor.start(outdir, resume=True, seed=None)
circuit2c_turtlebot_camera_dqn.py:            env.monitor.flush(force=True)
circuit2c_turtlebot_camera_dqn.py:                        env.monitor.flush()
circuit2c_turtlebot_camera_dqn.py:    env.monitor.close() #not needed in latest gym update
circuit2_turtlebot_lidar_dqn.py:        env.monitor.start(outdir, force=True, seed=None)
circuit2_turtlebot_lidar_dqn.py:        env.monitor.start(outdir, resume=True, seed=None)
circuit2_turtlebot_lidar_dqn.py:            env.monitor.flush(force=True)
circuit2_turtlebot_lidar_dqn.py:                        env.monitor.flush()
circuit2_turtlebot_lidar_dqn.py:    env.monitor.close()
circuit2_turtlebot_lidar_qlearn.py:    # env.monitor.start(outdir, force=True, seed=None)
circuit2_turtlebot_lidar_qlearn.py:            # env.monitor.flush(force=True)
circuit2_turtlebot_lidar_qlearn.py:    # env.monitor.close()
circuit2_turtlebot_lidar_sarsa.py:    env.monitor.start(outdir, force=True, seed=None)
circuit2_turtlebot_lidar_sarsa.py:            env.monitor.flush(force=True)
circuit2_turtlebot_lidar_sarsa.py:    env.monitor.close()
circuit_turtlebot_lidar_qlearn.py:    # env.monitor.start(outdir, force=True, seed=None)
circuit_turtlebot_lidar_qlearn.py:            # env.monitor.flush(force=True)
circuit_turtlebot_lidar_qlearn.py:    #env.monitor.close()
maze_turtlebot_lidar_qlearn.py:    env.monitor.start(outdir, force=True, seed=None)
maze_turtlebot_lidar_qlearn.py:            env.monitor.flush(force=True)
maze_turtlebot_lidar_qlearn.py:    env.monitor.close()
round_turtlebot_lidar_test.py:    env.monitor.start(outdir, force=True, seed=None)
round_turtlebot_lidar_test.py:            env.monitor.flush(force=True)
round_turtlebot_lidar_test.py:    env.monitor.close()
mkhansen@mkhansen-sky1:~/gym-gazebo/examples/turtlebot$

vmayoral, 2018-03-22:
@rkojcev how would you like us to proceed here?

PR #126 (2018-04-06): Fix monitor bug #120
Fixes #120 in all the examples. I could squash the changes into one commit if needed, I kept them separate so they could be reverted at the file level if needed.
closed 2018-04-06 by rkojcev