erlerobot/gym-gazebo #121: circuit2 DQN should have network_inputs = 20

mkhansen-intel, 2018-03-21:
When running the circuit2_turtlebot_lidar_dqn.py, the following error occurs:

  File "circuit2_turtlebot_lidar_dqn.py", line 340, in <module>
    qValues = deepQ.getQValues(observation)
  File "circuit2_turtlebot_lidar_dqn.py", line 148, in getQValues
    predicted = self.model.predict(state.reshape(1,len(state)))
  File "/usr/local/lib/python2.7/dist-packages/keras/models.py", line 1027, in predict
    steps=steps)
  File "/usr/local/lib/python2.7/dist-packages/keras/engine/training.py", line 1782, in predict
    check_batch_axis=False)
  File "/usr/local/lib/python2.7/dist-packages/keras/engine/training.py", line 120, in _standardize_input_data
    str(data_shape))
ValueError: Error when checking : expected dense_1_input to have shape (100,) but got array with shape (20,)

This can be fixed by changing circuit2_turtlebot_lidar_dqn.py line 288 to:
        network_inputs = 20

This is an easy fix, please implement.

mkhansen-intel, 2018-03-22:
OK, I see the issue now. I was running the other setup file: turtlebot_setup.bash. The difference appears to be that in the kobuki.urdf.xacro file for the 'nn' version it sets ray_count_horizontal = "200" instead of "40".   When I ran the right setup file, it now works. 

However, I'd like to understand why these are different. I understand that these have to do with the # of discrete inputs from the lidar. That means the DQN version is getting 100 inputs from the lidar and the Qlearn version is only getting 20. Is there a reason for that? Is it better to have 100 inputs to the DQN? I'm curious whether there is data to suggest that 100 inputs are needed?

mkhansen-intel, 2018-04-06:
@vmayoral @ahcorde - can anyone tell me why the DQN lidar input is set to 100 vs 20 for the DQN? Is there any benefit to that?

nzlz, 2019-01-15:
Sorry for the late reply @mkhansen-intel, you are right about the network inputs. We will take this into account if we decide to migrate the Turtlebot to the new version of gym-gazebo (which MARA-centered for now).

> That means the DQN version is getting 100 inputs from the lidar and the Qlearn version is only getting 20. Is there a reason for that?

The only reason is that during our tests we wanted to demonstrate that the DQN was able to handle better a larger input array than Q-learn.

> Is it better to have 100 inputs to the DQN?

As I said before, we just left it that way after experimenting a bit with it. I believe 20 inputs should be more than enough and would achieve the same level of performance, with the benefit that it would make the training a bit faster.

closed 2019-01-15 by nzlz

mkhansen-intel, 2019-01-15:
@nzlz - I haven't used this project in awhile, it seemed pretty abandoned at the time. It's nice to see someone actively working on it again. I had made a suggestion on Discourse that it be re-designed on ROS2 to be a library instead of a fork of AI Gym. Hopefully that will happen this year. I would love to see full support for some other robot arms and types of mobile robots too. But that's probably another Discourse thread.

mkhansen-intel, 2019-01-15:
@vmayoral - I apologize if I upset you. At the time I was using it there was no active development going on  and issues I filed often sat long periods of time before a response.  For example this issue that I filed in March was never assigned or answered until now, over 9 months later. I'm not complaining, just stating a fact. In fact my original comment was meant as a positive in that I'm glad to see someone actively working on it again.

nzlz, 2019-01-16:
Following your suggestion on Discourse @mkhansen-intel , the upcoming version of gym-gazebo is not a fork anymore. Instead it just uses `gym` as a library. We will be happy to receive more feedback once the work is published. 

Note that since we are fully focusing on ros2, we are now making an effort to migrate different packages to the new version. What I mean is we need to finish a few tasks before publishing the gym-gazebo, but won't take too long.

vmayoral, 2019-01-16:
@mkhansen-intel no hard feelings 👍, and frankly, I'm very happy seeing you jumping in and trying to help. This is what we've been hoping ever since we open sourced the project.

Cheers,

mkhansen-intel, 2019-01-16:
@nzlz - that is good news, I think you'll find that makes your life easier in the long run, and allows users like me to get the latest goodness from the gym package without requiring a rebase on your part except for API changes.

@vmayoral - I only wish I had more time to spend helping you all on this but I don't right now. I will be watching for updates however!