erlerobot/gym-gazebo #123: Is anyone actively maintaining and supporting this project?

mkhansen-intel, 2018-03-22:
@rkojcev, @vmayoral - Are you the maintainers of this project? Looking at the contributions list, it appears you are, but would like to verify. If so, are you actively maintaining it or is this an 'as-is' project?

The reason I ask is that I am very interested in contributing to it, but not sure if there is anyone reviewing PRs, working on CI, etc.

vmayoral, 2018-03-22:
Hello @mkhansen-intel,

The repository is active and *maintained* but mostly in our free time. This includes reviewing issues and PRs whenever we can find time for it. Some "community environments" are also reviewed from time to time. 

Besides ourselves and as far as we know, for now, `gym-gazebo` is mostly used in academia (research). There's been some recent growth in adoption recently but it's still very hard to work with it unless you know the technology behind it reasonably well. We have a few ideas on how to make the process simpler but it'll take some time until we can try them out.

Interest and contributions are welcome. Specially if they come from industry and with the backing of resources.

closed 2018-03-22 by vmayoral

mkhansen-intel, 2018-03-22:
@vmayoral - thank you for the fast response! I will talk with my management and get back to you.