erlerobot/gym-gazebo #124: rospy.wait_for_message breaks for depth image: Timeout error

zishanahmed08, 2018-03-26:
Hi,

I am always getting timeout errors related to the depth image which is published by gazebo. I tried to increase the timeout value, but it didn't helped.

Is there any advice on this issue?
>
        while image_data is None:
            try:
                image_data = rospy.wait_for_message('/camera/depth/image_raw', Image)
            except CvBridgeError as e:
                print(e)
The error is as follows:

File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/client.py", line 417, in wait_for_message
raise rospy.exceptions.ROSException("timeout exceeded while waiting for message on topic %s"%topic)
rospy.exceptions.ROSException: timeout exceeded while waiting for message on topic /camera/depth/image_raw
Thanks!

wyixiang, 2018-04-03:
I met the same problem, have you solved it?

zishanahmed08, 2018-04-03:
Unfortunately No

joellutz, 2018-05-03:
I have something like this, which should work good enough:
```
def read_kinect_point_cloud(self):
        data = None
        while data is None:
            try:
                data = rospy.wait_for_message('/camera/depth/points', PointCloud2, timeout=5)
            except:
                pass
        return data
```
Is your camera publishing this topic? You can check this with `rostopic list` (the simulation has to be running of course)