erlerobot/gym-gazebo #125: fail to use method in cv_bridge

wyixiang, 2018-04-03:
I changed some code to see the error, and the error is followed:

Traceback (most recent call last):
  File "circuit2c_turtlebot_camera_dqn.py", line 288, in <module>
    observation = env.reset()
  File "/home/yixiangw/gym-gazebo/gym_gazebo/envs/turtlebot/gazebo_circuit2c_turtlebot_camera_nn.py", line 217, in _reset
    cv_image = CvBridge().imgmsg_to_cv2(image_data, "bgr8")
  File "/home/yixiangw/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/vision_opencv/cv_bridge/python/cv_bridge/core.py", line 163, in imgmsg_to_cv2
    dtype, n_channels = self.encoding_to_dtype_with_channels(img_msg.encoding)
  File "/home/yixiangw/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/vision_opencv/cv_bridge/python/cv_bridge/core.py", line 99, in encoding_to_dtype_with_channels
    return self.cvtype2_to_dtype_with_channels(self.encoding_to_cvtype2(encoding))
  File "/home/yixiangw/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/vision_opencv/cv_bridge/python/cv_bridge/core.py", line 91, in encoding_to_cvtype2
    from cv_bridge.boost.cv_bridge_boost import getCvType
ImportError: dynamic module does not define module export function (PyInit_cv_bridge_boost)

so how can I solve it ?

zishanahmed08, 2018-04-03:
the error is related to cv2 issue with ROS.
Have you renamed cv2.so ,located at /opt/ros/kinetic/lib/python2.7/dist-packages/   to another name(eg:-cv2-renamed.so) ?

wyixiang, 2018-04-03:
Yes, and I have installed opencv for python3. That import cv2 is no problem, but I can't use the method in cv_birdge

zishanahmed08, 2018-04-03:
I guess. you need to compile them from source.Refer https://github.com/erlerobot/gym-gazebo/issues/90

Add 3 more packages to the file gazebo.repos locates in /home/zshaikh/projects/gym-gazebo/gym_gazebo/envs . thes packages were related to open cv for ros

geometry:
   type: git
   url: https://github.com/ros/geometry
   version: indigo-devel
 geometry2:
   type: git
   url: https://github.com/ros/geometry2
   version: indigo-devel
 vision_opencv:
   type: git
   url: https://github.com/ros-perception/vision_opencv.git
   version: kinetic

wyixiang, 2018-04-04:
and here are some messages for cv_birdge when I install this package

Scanning dependencies of target cv_bridge
Building CXX object vision_opencv-1.12.7/cv_bridge/src/CMakeFiles/cv_bridge.dir/cv_bridge.cpp.o
Building CXX object vision_opencv-1.12.7/cv_bridge/src/CMakeFiles/cv_bridge.dir/rgb_colors.cpp.o
Linking CXX shared library /home/yixiangw/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel/lib/libcv_bridge.so
Built target cv_bridge
Scanning dependencies of target cv_bridge_boost
Building CXX object vision_opencv-1.12.7/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/module.cpp.o
Building CXX object vision_opencv-1.12.7/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/module_opencv3.cpp.o
Linking CXX shared library /home/yixiangw/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel/lib/python2.7/dist-packages/cv_bridge/boost/cv_bridge_boost.so
Built target cv_bridge_boost

wyixiang, 2018-04-04:
It didn't work for me (in python3).
But I find an interesting thing is that cv_bridge can be run smoothly in python2, and I don't know why.
So I'm preparing the env for python2, if there's something new I will write here.

closed 2018-04-17 by wyixiang