erlerobot/gym-gazebo #127: turtlebot_gazebo not connecting /gazebo and turtlebot topics (odom, mobile_base)

salfityj, 2018-04-10:
Some preliminaries:
Ubuntu 14.04
ROS Kinetic
Python 2.7
Gazebo 8

TL;DR - Turtlebot isn't able to accept move commands because `/gazebo` and `/mobile_base` are not communicating properly.

Went through all the installation steps and I'm seeing a major issue. 
First off - to get this to build - I had to modify the `gazebo.repos` file
```
ar_track_alvar:
   type: git
   url: https://github.com/sniekum/ar_track_alvar
   version: kinetic-devel
...
ecl_core:
     type: git
     url: https://github.com/stonier/ecl_core
     version: release/0.61-indigo-kinetic
 ecl_lite:
    type: git
    url: https://github.com/stonier/ecl_lite
    version: release/0.61-indigo-kinetic
ecl_navigation:
    type: git
    url: https://github.com/stonier/ecl_navigation
    version: release/0.60-indigo-kinetic
ecl_tools:
    type: git
    url: https://github.com/stonier/ecl_tools
    version: release/0.61-indigo-kinetic
```
Everything built well after running `bash setup_kinetic.bash`

In all the examples in `/gym-gazebo/examples/turtlebot/`, when launched, the turtlebot is spawned in gazebo but there the turtlebot doesn't move. Upon diving deeper, I've found that the turtlebot (kobuki base) is not hooked up to gazebo properly. 

[image]

From `rqt_graph` output, you can see that the `/mobile_base` is not communicating well with `/gazebo`, meaning that the turtlebot won't be able to accept commands to move.

Examining an even simpler example, I'm trying to run the example `roslaunch turtlebot_gazebo turtlebot_world.launch` When doing a 1:1 comparison with another machine with Gazebo7, I'm noticing the `/gazebo` node not publishing important topics including `/odom` and some `/mobile_base` topics among others.

Here it is with Gazebo8
[image]

Here it is with Gazebo7
[image]

The difference is that the `/mobile_base` node is not communicating with `/gazebo`
I'm boiling this down to the file `/turtlebot_gazebo/launch/includes/kobuki.launch.xml` not setting things up like it should, but I can't find out what the problem is.

Has anyone got this combo (ubuntu 16.04, ROS Kinetic, Gazebo 8, Python 2.7) and did you go through any special steps? 

Thanks

JaxDraxler, 2018-04-13:
Hello !!
I don't think that ROS Kinetic works with Ubuntu 14.04

salfityj, 2018-04-13:
@JaxDraxler typo - thanks for catching my mistake !
Ubuntu 16*

JaxDraxler, 2018-04-19:
ar_track_alvar:
   type: git
   url: https://github.com/sniekum/ar_track_alvar
   version: kinetic-devel
...
ecl_core:
     type: git
     url: https://github.com/stonier/ecl_core
     version: release/0.61-indigo-kinetic
 ecl_lite:
    type: git
    url: https://github.com/stonier/ecl_lite
    version: release/0.61-indigo-kinetic
ecl_navigation:
    type: git
    url: https://github.com/stonier/ecl_navigation
    version: devel
ecl_tools:
    type: git
    url: https://github.com/stonier/ecl_tools
    version: release/0.61-indigo-kinetic

I guess you should leae the ecl_navigation version to devel. Because when i've changed it to release it raised an error message.