erlerobot/gym-gazebo #129: Turtlebot circles aimlessly

JaxDraxler, 2018-04-12:
Hello !!
I've installed ROS kinetic with Gazebo 8. I've run the setup_kinetic.bash, despite an error in the cmake the turtlebot still managed to boot and episodes started. 
However the turtle-bot seems to just circle aimlessly around itself until the episode terminates, not doing any intelligent action or learning.

Any help please ??

[image]

mkhansen-intel, 2018-04-13:
Are you running a gzclient? Which maze / map are you using? Can you submit a screenshot of the robot in the maze? I'll be honest I have no idea why the robot would go in circles, but more info might help.

JaxDraxler, 2018-04-13:
I was running the circuit2 qlearn example.
I have found the issue, it is an error in the setup_kinetic.bash.
You need to CATKIN_IGNORE the wiimote.
Thank you for responding.

closed 2018-04-13 by JaxDraxler

pulver22, 2018-11-06:
HI @JaxDraxler, did you solve? I have the same exact problem!