erlerobot/gym-gazebo #132: AttributeError: 'module' object has no attribute 'monitoring'

chanfoong, 2018-04-16:
I had started my training by `python circuit2_turtlebot_lidar_qlearn.py`, then when I try to run `python display_plot.py`. It came out this error
```
Traceback (most recent call last):
  File "display_plot.py", line 106, in <module>
    plotter.plot(full=True)
  File "display_plot.py", line 37, in plot
    results = gym.monitoring.load_results(self.outdir)
AttributeError: 'module' object has no attribute 'monitoring'

```
I'm also changed the line 37 in `display_plot.py` to `results = monitoring.load_results(self.outdir)`
It shows
```
Traceback (most recent call last):
  File "display_plot.py", line 106, in <module>
    plotter.plot(full=True)
  File "display_plot.py", line 37, in plot
    results = monitoring.load_results(self.outdir)
NameError: global name 'monitoring' is not defined
```
Does anyone have any solution to solve this?

mkhansen-intel, 2018-04-18:
@chanfoong - my PR #131 is now merged, if you re-pull the master branch you can try it.

chanfoong, 2018-04-20:
thanks @mkhansen-intel now the graph will pop out and update every 100 episode. BTW can we adjust the update episode like the graph will update every single episode? and how about the average reward graph(red line) and can we put average reward value in form of table?

sorry for the late respond but  I'm appreciate ur help

mkhansen-intel, 2018-04-20:
@chanfoong - all of those things are possible. Right now the display graph interval is hard-coded to 100 episodes, but you could easily change that in your copy to whatever you want. If I was to parameterize this, see issue #134 I would make it possible to change this as a parameter instead of hard-coded. However, I'm not sure I'll have time to do that any time soon. 

For the other changes, I'm sure it's possible, but not planning to do that myself. Feel free to do that yourself.

closed 2018-04-22 by chanfoong