erlerobot/gym-gazebo #133: DQN not running

JaxDraxler, 2018-04-16:
Hello !!
I've setup the bridge and the circuit2_turtlebot_lidar_qlearn.py is working perfectly. 
However, when i try to run the dqn file i get no error but i also get no output, so the program gets stuck like its waiting for something.

[image]

Any thoughts ??

mkhansen-intel, 2018-04-16:
@JaxDraxler - you are probably running into the error described here: (#122 ). This was fixed but a subsequent push to master over-wrote it so it needs to be fixed again. For now you could try that PR locally by doing this: 
```
git fetch origin pull/122/head:fix_dqn
git checkout fix_dqn
```
That will create a new local branch with the fix in it and checkout that branch. Hopefully this will help you in the short term until the issue is resolved in master branch again.

JaxDraxler, 2018-04-17:
Hello thank you for the fast response !! I tried the pull request put it didn't work. Is there a specific directory where it should have been done ?

JaxDraxler, 2018-04-17:
I'm getting an error about the monitor at first. I'll try and fix it, i think it is because the file is written in older python language not python3 maybe !!

[image]

mkhansen-intel, 2018-04-17:
@JaxDraxler - I fixed this in PR#130, which is now merged to master branch. You can get this change by re-pulling the master branch. If you run in to the other issue, I've submitted PR#132 to fix that, so you can pull that PR to get the latest working version. Let me know if you need more help.

JaxDraxler, 2018-04-18:
@mkhansen-intel  
Hello !! Thank you for the fix, I've also found an issue in the /gym-gazebo/examples/turtlebot/memory.py 
They were using an array at line 21 instead of a list, so the simulation would stop after two episodes.
After the fix i've got this:

[image]

mkhansen-intel, 2018-04-18:
@JaxDraxler  - What was the issue you found in memory.py, can you file a new issue on that and submit a PR to fix it?

JaxDraxler, 2018-04-19:
@mkhansen-intel Sure !!

closed 2018-04-19 by JaxDraxler