erlerobot/gym-gazebo #134: Proposal: Refactor and parameterize the examples

mkhansen-intel, 2018-04-16:
@vmayoral et al - I was thinking of refactoring all the examples to parameterize them, since they are all very similar yet contain many hard-coded values and duplicated code. I would ideally like there to be one gym_example.py that can be called with parameters to choose the robot type, maze, model type (qlearn or dqn), episode count, etc.. 

For example, a command line would be:
```
python gym_example.py -r turtlebot -m circuit2 -m qlearn -e 1000
```
Then the parameters could all map to values in a JSON config file that is read. So that the strings could be kept short and simple. 

This would remove a lot of code redundancy in the examples and make them easier to change or maintain.

Thoughts?

TheCamusean, 2018-04-17:
@mkhansen-intel - It is fine with me and not overlapping the proposal in #128 . 

Moreover, with a JSON config file , it would be quite easy to add new learning algorithms, enviroments or robots.

mkhansen-intel, 2019-01-15:
@vzlz - can this be done as part of the new version?

vmayoral, 2019-01-15:
@rkojcev, @YueErro, @alesolano what's your input here?

I think @nzlz could use your more experienced views here.

nzlz, 2019-01-16:
This was a good idea for gym-gazebo(1), but I don't know if it would make an impact in gym-gazebo(2). The idea relies on having many different `robots` and `worlds`, and for now, seems like the new version is not taking that path. Anyway its a good idea, which should be discussed once we take a final decision regarding the "targets" of the new version.

d3sm0, 2019-01-25:
Have you looked at [gin-config](https://github.com/google/gin-config). It turned out to be quite useful in my last rl project?

nzlz, 2019-01-25:
For now we have opted to give the user the ability to tune a few parameters from the command line. GUI/no GUI, sim speed real/ sim speed max, to name a couple. We will take your suggestion into account for the future @d3sm0 , thanks.