erlerobot/gym-gazebo #137: Population must ne a sequence or set

JaxDraxler, 2018-04-19:
Hello !!
When to try to run circuit2_turtlebot_lidar_dqn.py, it will raise an error: 
raise TypeError("Population must be a sequence or set.  For dicts, use
   list(d).")
TypeError: Population must be a sequence or set.  For dicts, use list(d).

This is due to an error in the code gym-gazebo/examples/turtlebot/memory.py
The following should fix it.

`import random
import numpy as np

class Memory:
    """
    This class provides an abstraction to store the [s, a, r, a'] elements of each iteration.
    Instead of using tuples (as other implementations do), the information is stored in lists 
    that get returned as another list of dictionaries with each key corresponding to either 
    "state", "action", "reward", "nextState" or "isFinal".
    """
    def __init__(self, size):
        self.size = size
        self.currentPosition = 0
        self.states = []
        self.actions = []
        self.rewards = []
        self.newStates = []
        self.finals = []

    def getMiniBatch(self, size) :
        indices = random.sample(list(np.arange(len(self.states))), min(size,len(self.states)) )
        miniBatch = []
        for index in indices:
            miniBatch.append({'state': self.states[index],'action': self.actions[index], 'reward': self.rewards[index]$
        return miniBatch

    def getCurrentSize(self) :
        return len(self.states)

    def getMemory(self, index): 
        return {'state': self.states[index],'action': self.actions[index], 'reward': self.rewards[index], 'newState': $

    def addMemory(self, state, action, reward, newState, isFinal) :
        if (self.currentPosition >= self.size - 1) :
            self.currentPosition = 0
        if (len(self.states) > self.size) :
            self.states[self.currentPosition] = state
            self.actions[self.currentPosition] = action
            self.rewards[self.currentPosition] = reward
            self.newStates[self.currentPosition] = newState
            self.finals[self.currentPosition] = isFinal
        else :
            self.states.append(state)
            self.actions.append(action)
            self.rewards.append(reward)
            self.newStates.append(newState)
            self.finals.append(isFinal)

        self.currentPosition += 1
`

mkhansen-intel, 2018-04-19:
@JaxDraxler - I haven't seen this error, does it happen every time you run?

JaxDraxler, 2018-04-20:
@mkhansen-intel Yes, every time i run DQN i got this error.
I've traced it back and it seems they are using random.sample in a wrong way.
it should be random.sample(list, k) but they are using it like this: random.sample(array, k)

mkhansen-intel, 2018-04-20:
@JaxDraxler - strange I haven't seen this, do you have to run for a long time to see it or does it happen right away?

JaxDraxler, 2018-04-21:
@mkhansen-intel yes it happens right away. It is when the DQN algorithm tries to access the memory code to start memorizing batches.

mkhansen-intel, 2018-04-23:
@JaxDraxler - can you give the exact sequence of commands and attach your terminal output? I don't see this problem at all. My guess is we're running different versions of Python or TensorFlow or AI Gym? It would be good to know which versions of packages you're running.

JaxDraxler, 2018-04-24:
@mkhansen-intel so this is the error before the fix. As you can see, it is related to the memory.py file that the DQN algorithm uses.

[image]

mkhansen-intel, 2018-04-24:
@JaxDraxler - it looks to me like you're calling learnOnMiniBatch after only 2 iterations. The code for calling it is this:
            if stepCounter >= learnStart:
                if stepCounter <= updateTargetNetwork:
                    deepQ.learnOnMiniBatch(minibatch_size, False)
                else :
                    deepQ.learnOnMiniBatch(minibatch_size, True)

So what do you have your "learnStart" variable set to? Mine is 64 by default (I haven't changed it). I think it needs to be the same as the minibatch size:
        minibatch_size = 64
        learnStart = 64

What do you have these set to?

JaxDraxler, 2018-04-25:
@mkhansen-intel both are also set to 64 by default in mine too. 

[image]

mkhansen-intel, 2018-04-25:
@JaxDraxler - I'm sorry I don't see that error at all and I'm running for 1000 epochs. Have you made any other changes or are you using the master branch of the repo? Have you made sure to pull the latest code from master branch and try that? If you still see the problem, then check your versions of AI Gym and numpy, there's some version mismatch somewhere. I don't know what else to suggest since I can't reproduce this.

JaxDraxler, 2018-04-25:
@mkhansen-intel Thank you for your help. I guess it has to do with the versions.

closed 2018-04-25 by JaxDraxler

mkhansen-intel, 2018-04-25:
@JaxDraxler - so were you able to resolve this? If so, please share what the fix of the issue was so that others can learn how to resolve it.

JaxDraxler, 2018-04-26:
Yes !! As i already cited, go to gym-gazebo/examples/turtlebot/memory.py
Then replace line 23 with : 
indices = random.sample(list(np.arange(len(self.states))), min(size,len(self.states)) )