erlerobot/gym-gazebo #138: Train using gym gazebo and test using own created gazebo track

chanfoong, 2018-04-21:
I'm training the turtlebot using gym gazebo and I'm thinking about can we save the **trained** turtlebot by some ways or save in something like trained weight file and put the trained turtlebot back into gazebo with different environment?

mkhansen-intel, 2018-04-23:
@chanfoong - yes this is possible. Are you using qlearn or dqn examples? If you're using dqn, it is already saving the weights every 100 episodes and all you have to do is re-load them with a different map. I have tried this, training using the circuit2 map and then loading in the maze map. I also have a patch to make it possible in qlearning too but would need to submit that in a PR if you need it.

chanfoong, 2018-04-24:
I'm doing with qlearning, can I do it with q learning and save it weight?

mkhansen-intel, 2018-04-27:
@chanfoong - did you try my code and if so how did it work for you?

chanfoong, 2018-05-09:
Hi @mkhansen-intel sorry for a really late reply because Im quite busy for past 2 weeks.
I straight a way go for a try on the qlearn method from ur branch. The code is working and robot is training, but I have few questions
1. Where can I find the weight file that saved? Is it any thing that related to "qlearn_states.npy" ? and can i review the file?
2. Does it have any way to know the weight had reloaded once do the qlearning training?

mkhansen-intel, 2018-05-09:
@chanfoong - 
1. yes, the file in my code is called qlearn_states.npy, but you can change that to whatever you want in your code. If I was to submit this I'd make it a parameter instead. It puts the file wherever the current working directory is. 
2. If it loads the file, you'll see a message printed to the screen "Loading from file: qlearn_states.npy"

Again, I expect this code to work, but I don't expect your experiment to work. I don't think the learning from one maze will transfer to another using qlearning. I have seen poor results when I tried that. I think you'll have better luck with the DQN models. Let me know how it works.

chanfoong, 2018-05-10:
@mkhansen-intel I had run the training but it seem like doesn't create the qlearn_states.npy in current directory. When I stop and run the training for second time, it doesn't show any message that "Loading from file: qlearn_states.npy" Is there any problem happen?

mkhansen-intel, 2018-05-10:
@chanfoong - can you paste in the commands you are running to start the program including what directories you are running them in? It places the qlearn_states.npy file in the directory from which the 'python circuit2_turtlebot_lidar_qlearn.py' command is run.

chanfoong, 2018-05-11:
@mkhansen-intel Im running by this command `~/gym-gazebo/examples/turtlebot$ python circuit2_turtlebot_lidar_qlearn.py`

mkhansen-intel, 2018-05-11:
@chanfoong - then it should be placing it in ~/gym-gazebo/examples/turtlebot

chanfoong, 2018-05-11:
@mkhansen-intel - The file is not appear in that directory

mkhansen-intel, 2018-05-11:
@chanfoong - here is a suggestion. For now, just modify the code to put a full path instead of "qlearn_states.npy". It can be something like "/temp/qlearn_states.npy". See if that works. If not, then add a print statement right before that section of code to make sure it is executing at all. You may have to debug this further. It works for me, so I'm not sure what the issue is. Let me know what you find out.

chanfoong, 2018-05-11:
@mkhansen-intel - I had try, I think the problem is on this two line `qlearn.loadQ(qfile)` `qlearn.saveQ(qfile)` because
1. I think the code didn't run through the save line because it didn't save any file in current directory.
2. I manually added a file named as 'qlearn_states.npy' in that directory and the code does detect it and print 'Loaded from file: qlearn_states.npy' but it come out error with

```
Traceback (most recent call last):
  File "suggest.py", line 48, in <module>
    qlearn.loadQ(qfile)
AttributeError: QLearn instance has no attribute 'loadQ'
```
That what I found, and does this problem can be done by `numpy.save(qfile)` or something like that??

nzlz, 2019-01-15:
@chanfoong if you are still interested in this matter please open a **Advice request** issue to facilitate issue-tracking. Also thanks @mkhansen-intel for so detailed explanations!

closed 2019-01-15 by nzlz