erlerobot/gym-gazebo #142: Turtlebot not spawned completly

JaxDraxler, 2018-04-25:
Hello !!

So i have a running gym-gazebo bridge but having a problem when opening gzclient.

I can only see the LIDAR and not the turtlebot !!

[image]

There is no raised error or issue. Any ideas on what might be going on ?

Thank you

pmwenzel, 2018-04-26:
Hi,

did you make sure that you have all the turtlebot/turtlebot_create/turtlebot_simulator packages in your catkin workspace? This can be an issue. What ROS/Gazebo configuration are you using?

JaxDraxler, 2018-04-26:
@pmwenzel Hello !!

I am using ROS kinetic with gazebo 8.4

Yes i think all packages are present in my workspace if i am not mistaking !!

[image]

geerfing, 2018-05-02:
Hi,
I have a running  gym-gazebo ,but it didn't appear the environment,how should I do .
Thank you!

JaxDraxler, 2018-05-08:
@geerfing Hello!! There should be a raised error somewhere when you run the code that might help figure out what is the problem

closed 2018-05-12 by JaxDraxler