erlerobot/gym-gazebo #146: Circuit2_turtlebot_lidar_dqn.py stuck and doesn't run

aunsid, 2018-05-01:
Whenever I try running this example, the terminal shows:
[image]

and nothing happens even in the gzclient the turtlebot is stuck in one position.

mkhansen-intel, 2018-05-01:
Hi @aunsid. Can you show what commands you ran (turtlebot_setup.bash, etc) and copy & paste the info from your screen from the last command you typed to this point above. It is likely there is an error message that already scrolled off-screen above your screenshot. If you can give more info, I'll try to help. You shouldn't have to run for 60 hours. You should be able to see progress in the form of Episode summaries right away.

aunsid, 2018-05-01:
[image]
[image]
[image]
[image]

And another problem that I have faced while running other algorithms search as q-learn, I either get an error saying something about env.reset() or one of the sensors stops working. It has never been able to run all 10,000 episodes. 3000 at max.
Thanks!

geerfing, 2018-05-02:
I  have  the same question, I still don't  solve it now

aunsid, 2018-05-03:
Could you please make a README file for python2 ?

closed 2018-05-03 by aunsid

zacwellmer, 2018-05-11:
@aunsid Did you end up sorting this out?

aunsid, 2018-05-11:
Well yeah, by installing it on on python2. Although I havent been able to complete 1000 epochs cause i get some kind of an error stating that one of the sensors took too long to respond, resulting in a core dump.

aunsid, 2018-05-11:
@mkhansen-intel does the camera_dqn.py work for you?

zacwellmer, 2018-05-11:
yeah, that is exactly the problem I am running into with the circuit_turtlebot_lidar_qlearn.py script. @aunsid  issue #148 w/ instructions is how I managed to get to this point.

Below is the core dump I'm running into(sounds the same as yours)
```
gzserver: /build/gazebo-nhSAPd/gazebo-7.0.0+dfsg/gazebo/sensors/SensorManager.cc:545: void gazebo::sensors::SensorManager::SensorContainer::RunLoop(): Assertion `(diffTime.sec < 1)&&("Took over 1.0 seconds to update a sensor.")' failed.
Aborted (core dumped)
[gazebo-1] process has died [pid 9240, exit code 134, cmd /home/zacwellmer/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -e ode /home/zacwellmer/gym-gazebo/gym_gazebo/envs/installation/../assets/worlds/circuit.world __name:=gazebo __log:=/home/zacwellmer/.ros/log/7cca1d54-54c7-11e8-b9d4-000c299ea188/gazebo-1.log].
log file: /home/zacwellmer/.ros/log/7cca1d54-54c7-11e8-b9d4-000c299ea188/gazebo-1*.log
```

aunsid, 2018-05-11:
I was able to get qlearn to work just fine. Try running it circuit_turtlebot_lidar_qlearn.py  -j(no of cores)
so if you have 6 cores it'll be circuit_turtlebot_lidar_qlearn.py -j6. Have the system monitor on, and DONT run gzclient in another terminal. I think its a processor issue. Cause I stopped all my other tasks and was able to get to more episodes with no other processes running in the background.

zacwellmer, 2018-05-11:
still crashes on me. Perhaps it's because I'm using a VM? But I give the VM 3 cores and only specify using 1 core to the circuit_turtlebot_lidar_qlearn.py script

aunsid, 2018-05-11:
I was using Intel  Xeon(R) CPU E3-1271, it had eight cores.

mkhansen-intel, 2018-05-11:
@aunsid - I haven't tried running the 'camera' version.

ElizabethDuo, 2018-08-16:
@zacwellmer Hi Zac! I also met the core dump problem when running `python circuit_turtlebot_lidar_qlearn.py `. Have you solved this problem? Any good ideas?