erlerobot/gym-gazebo #150: strangely unbale to register the node when i run the circuit2_turtlebot_lidar_dqn.py

zengjie617789, 2018-05-09:
i meet the problem when i run my rivised code,i thought i was because of my own code ; but when i git pull the new code and run it again ,it meet the same problem, anyone who can help me ,please!
i doubt if my keras gpu config is not correct. here are  my screenshots blow:
 

[image]

[image]

mkhansen-intel, 2018-05-10:
@zengjie617789 - if you open another window and run 'top' do you see the rosmaster node running? It looks to me like it has died somehow.

zengjie617789, 2018-05-11:
@mkhansen-intel  yes,it's still running 
[image]

mkhansen-intel, 2018-05-11:
@zengjie617789 - are you running turtlebot_setup_nn.bash first? Are you getting the error at the same time every time you run? What exact commands are you running to start the example? Have you tried without your modified code first and does that work?

zengjie617789, 2018-05-12:
@mkhansen-intel  yes ,i run it without my modified code ,but it still got the same problem. I have run turtlebot_setup_nn.bash first. when i enter the floder,i found the file "turtlebot_cnn_setup.bash", should i run it ?

nzlz, 2019-01-15:
@zengjie617789  `turtlebot_cnn_setup.bash` sets the corresponding environment variables to work with the turtlebot CNN example, not NN (not using the convolutional network required to work with vision). If you are still interested in this matter please open a **Advice request** issue to facilitate issue-tracking.

closed 2019-01-15 by nzlz