erlerobot/gym-gazebo #154: The turtlebot  didn't move

geerfing, 2018-05-15:
Hello! I run the command :python3 circuit2c_turtlebot_camera_dqn.py
The  turtlebot  didn't move,how should I  solve it? It appears  like  this:
[image]
[image]

radisc, 2018-08-22:
having the same problem

pulver22, 2018-11-07:
Same here! Did anyone solve it?

nzlz, 2018-11-07:
I remember we had this same issue when we implemented this solution and we managed to fix it. I will try to look at it when I have time because right now I can't remember, but for sure we did not use python3. Also make sure to perform a clean installation. @IkerZamora any idea?

nzlz, 2019-01-15:
Issue related to the usage of python3. Please stick to python2 or fill a **Feature request** issue to facilitate issue-tracking. However, note that right now there are not plans to enable python3 for the turtlebot.

closed 2019-01-15 by nzlz