erlerobot/gym-gazebo #156: X_batch = np.empty((0,self.input_size), dtype = np.float64) what is the meaning of 0

geerfing, 2018-05-23:
[image]
I want to know why the number is 0,np.empty((0,self.input_size), dtype = np.float64) 
[image]
Thanks!

zengjie617789, 2018-05-24:
～generate a matrix which of row is one and column is input_size

zengjie617789, 2018-05-24:
@geerfing yes i think you are right ,but this make me think what is the function of "X_batch"?

nzlz, 2019-01-15:
If you are still interested in this matter, please fill an **Advice request** issue to facilitate issue-tracking.

closed 2019-01-15 by nzlz