erlerobot/gym-gazebo #159: Invalid <arg> tag: turtlebot_description

TPEmist, 2018-06-15:
Since I'm trying to run the example > python3 circuit2_turtlebot_lidar_qlearn.py

But coming up with the error:
----
while processing /home/tpemist/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/turtlebot_gazebo/launch/includes/kobuki.launch.xml:
Invalid <arg> tag: turtlebot_description
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/tpemist/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share. 

Arg xml is <arg default="$(find xacro)/xacro.py '$(find turtlebot_description)/robots/$(arg base)_$(arg stacks)_$(arg 3d_sensor).urdf.xacro'" name="urdf_file"/>
The traceback for the exception was written to the log file

>>
.
.
_### and also try this solution below:
https://github.com/erlerobot/gym-gazebo/issues/58
but still not work
please help_

zengjie617789, 2018-07-02:
you should source your workplace or add the path into the ~/.bashrc

nzlz, 2019-01-15:
Hi @TPEmist sorry for the late reply! You should be using python2. We are not planning to migrate the Turtlebot to python3, but if you want to share your thoughts or any other kind of problem, please fill a **Feature resquest** of **Bug request** issue to facilitate issue-tracking.

closed 2019-01-15 by nzlz