erlerobot/gym-gazebo #161: "circuit2_turtlebot_lidar_qlearn.py"can not run UNDER Pycharm

zengjie617789, 2018-07-01:
when i run the py file UNDER Pycharm , it shows the error "FileNotFoundError: [Errno 2] No such file or directory: 'roscore'"

1.it can run under the terminal successfully，
[image]
 but meet erros under Pycharm:
[image]
2.the path of virtuanenv in Pycharm is 
[image]
anyone who can help ? 
thank you in advance~

pmwenzel, 2018-07-02:
You should make sure that you run PyCharm from the terminal to make sure your .bashrc is initialised correctly, hence sourcing your ROS environment. Hope that helps.

zengjie617789, 2018-07-03:
@pmwenzel   I did what you said. I follow the advice that to add something in the Jetbranis-pycharm.desktop; as show blow :   
[image]
the file of .bashrc  was :  
[image]
 and  what puzzled me is i can import rospy successfully in the Pycharm~
 
could you give me some other advice? thank you~

pmwenzel, 2018-07-06:
I don't know either. Maybe some of your ROS dependencies are screwed up or do you only get the errors when running from PyCharm?

zengjie617789, 2018-07-08:
the Environment Variables make me mad ~~~~~~~~~~~~

[image]

nzlz, 2019-01-15:
Hi @zengjie617789 , if you still need help with this matter please fill a **Advice request** issue to facilitaty issue-tracking.

closed 2019-01-15 by nzlz