erlerobot/gym-gazebo #162: Ubuntu 16.04 Python Issue (ver 2 vs 3)

srkamo, 2018-07-02:
I am having an issue with my Python versions. ROS uses python 2 but the qlearn python script uses python 3 and I receive a syntax error when I try to learn the sample file.

here is the output:

`srkamo@mypc:~/gym-gazebo/examples/turtlebot$ roslaunch turtlebot-
Traceback (most recent call last):
  File "/opt/ros/kinetic/bin/roslaunch", line 34, in <module>
    import roslaunch
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 51, in <module>
    from . import param_dump as roslaunch_param_dump
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/param_dump.py", line 40, in <module>
    import roslaunch.config
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 45, in <module>
    import rospkg.distro
  File "/home/SR31523/anaconda3/lib/python3.6/site-packages/rospkg/distro.py", line 44, in <module>
    import yaml
  File "/home/SR31523/anaconda3/lib/python3.6/site-packages/yaml/__init__.py", line 284
    class YAMLObject(metaclass=YAMLObjectMetaclass):
                              ^
SyntaxError: invalid syntax
`

ElizabethDuo, 2018-08-07:
Maybe you can find solution in  #90 .
Or you can create a virtualenv and use python3 in that env, which is also explained how in issue90.
Good luck :)

nzlz, 2019-01-15:
@srkamo if you are still interested in this please fill a **Feature request** issue to facilitate issue-tracking.

closed 2019-01-15 by nzlz