erlerobot/gym-gazebo #163: tagged gym-gazebo repo to work with Dockerfile

mwagyuvm, 2018-07-04:
I'm trying to build the docker image (docker build -t gym-gazebo .) but run into an issue in step 24, "RUN cd ~/ros_catkin_ws && wstool init -j1 src kinetic-ros_comm-wet.rosinstall" as follows.

```
Step 24/125 : RUN cd ~/ros_catkin_ws && wstool init -j1 src kinetic-ros_comm-wet.rosinstall
 ---> Running in 0489b1a426b5
Using initial elements from: kinetic-ros_comm-wet.rosinstall
Writing /root/ros_catkin_ws/src/.rosinstall
ERROR [vcstools] Tarball download unpack failed: file could not be opened successfully[/vcstools]
ERROR [vcstools] Tarball download unpack failed: file could not be opened successfully[/vcstools]
ERROR [vcstools] Tarball download unpack failed: file could not be opened successfully[/vcstools]
ERROR [vcstools] Tarball download unpack failed: file could not be opened successfully[/vcstools]
```
I don't see any tags in this repo; but are there points at which the gym-gazebo has been tested to have successful docker builds so I can work from those and have a working gym-gazebo container up quickly?

AustinDeric, 2018-07-16:
@mwagyuvm I can confirm that the Dockerfiles are not building.  I am working on an update, should be ready this week.  see issue #96

mwagyuvm, 2018-07-22:
Very good news. Please let me know if I can help in any way.

nzlz, 2019-01-15:
Lets follow this at https://github.com/erlerobot/gym-gazebo/issues/96 .

closed 2019-01-15 by nzlz