erlerobot/gym-gazebo #166: How can I set a navigation goal in maze environment for turtlebot?

ElizabethDuo, 2018-07-19:
I want to add a navigation goal in maze environment and change unoriented  walk to oriented navigation. Any good ideas to do that? How can I publish navigation target msgs and show the navigation goal in maze?
Many thanks!!

pmwenzel, 2018-07-23:
You can simply set a x and y coordinate corresponding to your goal location. 

Use the following subscriber to get the current x and y position of the robot:
```
rospy.Subscriber(name='/gazebo/model_states', data_class=ModelStates,
                                             callback=self.position_callback)
```

Either you are working on RL, you should adopt your reward function accordingly or you use a planner like A*. Hope that helps. You can visually set an object to your goal location in gazebo if you want to.

yangchen12, 2018-11-20:
> You can simply set a x and y coordinate corresponding to your goal location.
>
> Use the following subscriber to get the current x and y position of the robot:
>
> ``` [...]

Hello,how can konw if the robot reaceh the goal by the code?

pmwenzel, 2018-11-20:
Just check the squared distance between your goal point and the actual location of the robot and compare it with some threshold. This should be the easiest way.

yangchen12, 2018-11-20:
> Just check the squared distance between your goal point and the actual location of the robot and compare it with some threshold. This should be the easiest way.
Can you help with the code?Many thanks!!

yangchen12, 2018-11-21:
> Hi @yangchen12 ,
> 
> I believe you should change the `_step(..) `function in your environment so that it returns the boolean final state if the robot reaches the position (done/ not done). @pmwenzel has already mentioned a good way to calculate if the goal position if correct or not.
> 
> About the specific code, a quick google search will provide you with the code you need. Please share your own code if you have more specific issues.

Many thanks!!If I have questions,I will ask you for your help.Many thanks!!

closed 2018-12-29 by nzlz