erlerobot/gym-gazebo #168: ModuleNotFoundError: No module named 'std_msgs'

pallavijog13, 2018-08-07:
I am trying to install gym_gazebo on my Ubuntu 16.04 LTS system.

Everything is getting installed correctly, however, while trying to run **python circuit2_turtlebot_lidar_qlearn.py** command, I get error as

`ModuleNotFoundError: No module named 'std_msgs'.`

This should be part of **ros_comm** project which I have installed but still I cannot see std_msgs related code file anywhere.

Can someone please point me to how can I install std_msgs, so that I can get rid of this error message?

Thanks in advance!

ElizabethDuo, 2018-08-07:
Try download std_msgs package from [https://github.com/ros/std_msgs.git](url) and redo bash setup_kinetic.bash `.
Maybe it is simplu because of you don't have the std_msgs module.

pjog, 2018-08-08:
Thanks! Sorry I forgot to update. I could manage to install from above repository and also executed bash command as you have mentioned. Now, I have std_msgs folder and the code can recognize it. However, now I am getting another error for line **from std_msgs.msg import Header** as - 

`ImportError: cannot import name 'Header'`

I checked my folder "std_msgs/msg" and found several files with .msg extension and among them there is a Header.msg file. However, since it is not .PY file, import command does not recognize it. Is there any other step to convert these files to .PY file?

closed 2018-09-20 by rkojcev