erlerobot/gym-gazebo #171: MPI support for parallel execution?

joellutz, 2018-08-16:
Is it possible to run a gym-gazebo environment with MPI for parallel execution?

I've taken the [baseline DDPG implementation](https://github.com/openai/baselines/tree/master/baselines/ddpg) and adapted it to my own gym-gazebo environment. As the env.step() method is very slow (MoveIt takes forever to move my robot arm), I would like to run certain things in parallel on my ~16 CPU cores, e.g. the env.step() method or even the whole training loop.

Unfortunately, when running the DDPG implementation with my gym-gazebo environment and MPI (`mpirun -n 8 python myscript.py`) it doesn't work: It tries to start multiple ROS cores and Gazebo servers, which isn't possible. Is there a way to resolve this?

pulver22, 2018-11-06:
Hi @joellutz, were you able to solve your problem? I would be really interested in applying the same idea.

joellutz, 2018-11-06:
Hi @pulver22, unfortunately I wasn't able to solve this problem. Well I didn't really try too hard to solve it, as I had other problems with Gazebo crashing and far too slow MoveIt-execution anyway. You would want to have multiple ROS-cores and Gazebo servers running to have true parallelism, which won't be easy to do (if not impossible).

nzlz, 2019-01-15:
You can execute as many parallel executions as you like since now the main gazebo environment takes care of creating a new ROS_MASTER_URI and GAZEBO_MASTER_URI. The difficulty will come when you have to merge the learnt policies. 

If you still require information about this topic, please please fill an **Advice request** issue to facilitate issue-tracking.

closed 2019-01-15 by nzlz