erlerobot/gym-gazebo #173: Is it possible to work it in conda environrment?

Geonhee-LEE, 2018-08-27:
Hello.

I have wanted to adapt reinforcement learning to the GAZEBO.

However, When I follow your [INSTALL.md](https://github.com/erlerobot/gym-gazebo/blob/master/INSTALL.md), It didn't work.

My develop environment is Ubuntu16.04 and ros-kinetic.

I create the virtual environment using Conda like this conda create -n tensorflow pip python=3.5.1

I installed every process such as Theano, gym, Keras..

But, when I commanded `bash setup_kinetic.bash`, It had error...

Who you know what it is problems?		

I really want to use gym-gazebo.

Thanks you for your interest.

ahcorde, 2018-08-27:
Unfortunately installing `gym-gazebo` is not a straightforward process. If you post your errors we can try to help you.

Geonhee-LEE, 2018-08-27:
Thanks @ahcorde  
I just uploaded the picture.

[image]

when I commanded `bash setup_kinetic.bash` in the installation folder, I saw "Package "mav_msgs" not found in the workspace

ahcorde, 2018-08-27:
Look at this: `Packages "mav_msgs" not found in the workspace`

You need to include this [package](https://github.com/ethz-asl/mav_comm) into your workspace

closed 2018-08-29 by Geonhee-LEE