erlerobot/gym-gazebo #175: Error when running circuit2_turtlebot_lidar_qlearn.py

arthur960304, 2018-09-18:
When I tried this command `python3 circuit2_turtlebot_lidar_qlearn.py`
I got the following error but I cannot figure it out the reason

However when I try `gzclient` I could see turtlebot moving extremely slow in the map
but I don't think this is the proper result.

Any help would be really appreciated!
Thanks!

Here is the error:
[image]

Here is the view in gzclient:
[image]

MohmadAyman, 2018-09-20:
Did you try reinstalling `ros-kinetic-urdf` ?

arthur960304, 2018-10-02:
Hi @MohmadAyman  Yes I did, but the issue is still there. Any other suggestions?

MohmadAyman, 2018-10-02:
The dynamic linking error usually caused by some version mismatching, probably between `urdf`  and `robot_state_publisher` packages.

nzlz, 2019-01-15:
Although some users managed to train the Turtlebot in using python**3**, it is not officially supported since it is known to cause problems. If you are specially interested in this feature please open a **Feature request** issue to facilitate issue-tracking.

closed 2019-01-15 by nzlz