erlerobot/gym-gazebo #178: ImportError: dynamic module does not define init function (init_tf2)

andjsmile, 2018-10-11:
python circuit2_turtlebot_lidar_qlearn.py 
... logging to /home/parallels/.ros/log/e7cb8e46-cd06-11e8-b99c-001c42d76824/roslaunch-parallels-vm-14519.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://parallels-vm:36113/
ros_comm version 1.12.14

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES

roscore cannot run as another roscore/master is already running. 
Please kill other roscore/master processes before relaunching.
The ROS_MASTER_URI is http://parallels-vm:11311/
The traceback for the exception was written to the log file
Roscore launched!
Gazebo launched!
... logging to /home/parallels/.ros/log/e7cb8e46-cd06-11e8-b99c-001c42d76824/roslaunch-parallels-vm-14542.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://parallels-vm:44173/

SUMMARY
========

PARAMETERS
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /cmd_vel_mux/yaml_cfg_file: /home/parallels/g...
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 30.0
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True

NODES
  /
    bumper2pointcloud (nodelet/nodelet)
    cmd_vel_mux (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    mobile_base_nodelet_manager (nodelet/nodelet)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawn_turtlebot_model (gazebo_ros/spawn_model)

ROS_MASTER_URI=http://parallels-vm:11311/

process[gazebo-1]: started with pid [14582]
process[spawn_turtlebot_model-2]: started with pid [14586]
process[mobile_base_nodelet_manager-3]: started with pid [14587]
process[cmd_vel_mux-4]: started with pid [14588]
process[bumper2pointcloud-5]: started with pid [14596]
process[robot_state_publisher-6]: started with pid [14609]
Traceback (most recent call last):
  File "/home/parallels/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/spawn_model", line 32, in <module>
    import tf.transformations as tft
  File "/home/parallels/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel/lib/python3/dist-packages/tf/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 30, in <module>
  File "/home/parallels/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel/lib/python3/dist-packages/tf2_ros/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 38, in <module>
  File "/home/parallels/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel/lib/python3/dist-packages/tf2_py/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 38, in <module>
ImportError: dynamic module does not define init function (init_tf2)
[ INFO] [1539241271.823557771]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1539241271.824836465]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[spawn_turtlebot_model-2] process has died [pid 14586, exit code 1, cmd /home/parallels/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/spawn_model -unpause -urdf -param robot_description -model mobile_base __name:=spawn_turtlebot_model __log:=/home/parallels/.ros/log/e7cb8e46-cd06-11e8-b99c-001c42d76824/spawn_turtlebot_model-2.log].
log file: /home/parallels/.ros/log/e7cb8e46-cd06-11e8-b99c-001c42d76824/spawn_turtlebot_model-2*.log
[ INFO] [1539241272.314506906, 0.140000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1539241272.359788223, 1.440000000]: Physics dynamic reconfigure ready.

Please help me, thanks!

nzlz, 2019-01-15:
@srikanth-kilaru you won't face this issue in later versions of gym-gazebo since now each execution creates its own network segmentation, meaning it will have its own ROS_MASTER_URI.
If you are still facing this or a similar problem please fill a **Bug report** issue to facilitate issue-tracking.

closed 2019-01-15 by nzlz

XuboLyu, 2019-01-21:
For extra information, I use python3, and I rebuild "tf" package with catkin build for python3 from [here](https://github.com/ros/geometry2/issues/259). Since original "tf" just works for python2.