erlerobot/gym-gazebo #180: Installation error during "bash setup_kinetic.bash"

srikanth-kilaru, 2018-10-18:
Hi
I am getting the following error even after following all the steps. Can you please help
bash: /home/srikanth/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel/setup.bash: No such file or directory
Thanks

shulace13, 2018-10-19:
I had the similar problem too.
You need to go to that catkin_ws directory and catkin_make first, before you get all the repositories provided by the setup_kinetic.bash
[This page](http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment) should help out.
And after you do that, you will need to install some more packages to get the catkin_make work out.'

shulace13, 2018-11-06:
Hi @srikanth-kilaru 
Did you solve your problem?

srikanth-kilaru, 2018-11-06:
Hi
No I did not try it since as something regarding the installing messed up
my regular RoS environment and I had to back up

shulace13, 2018-11-06:
@srikanth-kilaru 
Oh no, sorry to hear...
Hope everything works out!

nzlz, 2019-01-15:
@srikanth-kilaru if you are still facing this problem please fill a **Bug report** issue to facilitate issue-tracking.

closed 2019-01-15 by nzlz