erlerobot/gym-gazebo #181: Turtlebot spins on the spot

pulver22, 2018-11-07:
Hello, 

I am trying to run the examples, in particular both the `circuit_turtlebot_lidar_qlearn.py` and the `circuit2_turtlebot_lidar_qleanr.py`. 

The behaviour I see is the same reported [here](https://github.com/erlerobot/gym-gazebo/issues/49#issuecomment-349249929). Basically, the simulation continuously stops and resumes (possibly for allowing taking the observation and then send an action) and the robots mainly rotates on the spot really slowly. Listening to the velocity topic I noticed the messages go through (linear velocity on x and also angular on z).

I also noticed the real time factor of gazebo is around 55 meaning the simulation is not real time (good idea for speeding up but I cannot explain why the robot moves so slowly).

Another point I noticed is that, after an episode starts and the reset() method is called, the robot is respawn always in the same position.

I run the simulation for 15 hours and I didn't see any learning at all. Is all this expected? What is the robot supposed to do/learn?

XuboLyu, 2019-01-13:
@vmayoral  same issue when running circuit_turtlebot_lidar_qlearn.py. It seems that the robot just moving around the same spot. Also observe the reward, it seems the robot does not learn at all. Who knows how to fix this?

nzlz, 2019-01-13:
Hi @XuboLyu , can you confirm that the same issue is hapenning with cirtuit2_turtlebotlidar_qlearn ? Also, which Ubuntu/ros versions are you using? Are you running the script with **python2**?

Check [this video](https://www.youtube.com/watch?v=8hxCBkgp95k) to see the expected behavior.

XuboLyu, 2019-01-13:
@nzlz  Yes. I can be sure about this. it's totally different with the expected behavior you give. Also from gzclient, I can only see a robot moving around same place (look like the original point) without any other objects(walls, for example). 

Yes, I use python2 to run the script and I use ubuntu 16.04 with ros kinetic 1.12.14

The reward is like always around -200: 
EP: 843 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.28] - Reward: -194     Time: 0:05:30
EP: 844 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.28] - Reward: -198     Time: 0:05:31
EP: 845 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.28] - Reward: -200     Time: 0:05:31
EP: 846 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.28] - Reward: -194     Time: 0:05:32
EP: 847 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -200     Time: 0:05:32
EP: 848 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -199     Time: 0:05:32
EP: 849 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -200     Time: 0:05:32
EP: 850 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -200     Time: 0:05:33
EP: 851 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -200     Time: 0:05:33
EP: 852 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -200     Time: 0:05:33
EP: 853 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -193     Time: 0:05:34
EP: 854 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -200     Time: 0:05:34
EP: 855 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -194     Time: 0:05:34
EP: 856 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -199     Time: 0:05:35
EP: 857 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -199     Time: 0:05:35
EP: 858 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -200     Time: 0:05:35
EP: 859 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -200     Time: 0:05:35
EP: 860 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -199     Time: 0:05:36
EP: 861 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -200     Time: 0:05:36
EP: 862 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -198     Time: 0:05:36
EP: 863 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -199     Time: 0:05:37
EP: 864 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -200     Time: 0:05:37
EP: 865 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -200     Time: 0:05:37
EP: 866 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -200     Time: 0:05:37
EP: 867 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -200     Time: 0:05:38
EP: 868 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -194     Time: 0:05:38
EP: 869 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.27] - Reward: -200     Time: 0:05:38

XuboLyu, 2019-01-13:
@nzlz  

Sure. Below is the screenshot. 
[image]

But new issue comes, when I try to rerun the script for screenshot, it stuck long time at the bold line. Then each episode takes much longer time than before. Everything is unchanged, but when run the scripts many time, issues above happen.
[ INFO] [1547351188.068617766]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1547351188.069140039]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1547351188.653673889, 0.023000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1547351189.002485989, 0.147000000]: Camera Plugin (robotNamespace = /), Info: Using the 'robotNamespace' param: '/'
[ INFO] [1547351189.007057869, 0.147000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1547351189.060869345, 0.147000000]: Laser Plugin (robotNamespace = /), Info: Using the 'robotNamespace' param: '/'
[ INFO] [1547351189.060912963, 0.147000000]: Starting Laser Plugin (ns = /)!
[ INFO] [1547351189.063116448, 0.147000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1547351189.074331984, 0.147000000]: Physics dynamic reconfigure ready.
[ INFO] [1547351189.165668116, 0.147000000]: Starting plugin Kobuki(ns = //)!
[ WARN] [1547351189.165719355, 0.147000000]: Kobuki(ns = //): missing <rosDebugLevel> default is na
[ INFO] [1547351189.166819946, 0.147000000]: Kobuki(ns = //): <tf_prefix> = 
[ INFO] [1547351189.166969248, 0.147000000]: Will publish tf. [mobile_base]
[ INFO] [1547351189.169192336, 0.147000000]: Kobuki(ns = //): Advertise joint_states[joint_states]!
[ INFO] [1547351189.170122076, 0.147000000]: Kobuki(ns = //): Advertise Odometry[odom]!
[ INFO] [1547351189.175689608, 0.147000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
[ INFO] [1547351189.181784019, 0.147000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/reset_odometry!
[ INFO] [1547351189.187819135, 0.147000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
[ INFO] [1547351189.188752346, 0.147000000]: Kobuki(ns = //): Advertise Cliff[mobile_base/events/cliff]!
[ INFO] [1547351189.189623823, 0.147000000]: Kobuki(ns = //): Advertise Bumper[mobile_base/events/bumper]!
[ INFO] [1547351189.190543900, 0.147000000]: Kobuki(ns = //): Advertise IMU[mobile_base/sensors/imu_data]!
[ INFO] [1547351189.190561154, 0.147000000]: GazeboRosKobuki plugin ready to go! [mobile_base]
[spawn_turtlebot_model-3] process has finished cleanly

### log file: /home/xlv/.ros/log/cdebe722-16e5-11e9-9591-b06ebfbb4fe5/spawn_turtlebot_model-3*.log

/usr/lib/python2.7/dist-packages/matplotlib/backend_bases.py:2437: MatplotlibDeprecationWarning: Using default event loop until function specific to this GUI is implemented
  warnings.warn(str, mplDeprecation)
EP: 1 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.9] - Reward: 1000     Time: 0:01:44
EP: 2 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.9] - Reward: 1000     Time: 0:03:27

nzlz, 2019-01-13:
Seems very strange to me that not even the circuit is showing up, did you run:
```bash
cd gym_gazebo/envs/installation
bash turtlebot_setup.bash
```

XuboLyu, 2019-01-13:
@nzlz  

Ohhhhhh. Thank you so much. It works as you did. I totally forget ...... Thanks a loooot!