erlerobot/gym-gazebo #183: No module named gym_gazebo

taihirose, 2018-11-13:
I have a same problem(https://github.com/erlerobot/gym-gazebo/issues/152).
I can't slove this. Sorry my poor English. Please tell me.

[image]

nzlz, 2018-11-14:
Hi @taihirose,
First thing, i recommend you go back to your home directory `cd ~` before you clone the gym-gazebo repo in [this ](https://github.com/erlerobot/gym-gazebo#running-an-environment)step, and keep following the installation steps from that point. Also note that gym and gym-gazebo are different packages.

taihirose, 2018-11-15:
Thank you for your answer!!
But same error display.
Do you have any solutions?

taihirose, 2018-11-15:
I have reinstalled,and this error display.What does this mean?
/home/hirose/ピクチャ/Screenshot from 2018-11-15 19-25-52.png

taihirose, 2018-11-15:
I have reinstalled,and new error display.What does this mean?
[image]

taihirose, 2018-11-15:
This Usage step(https://github.com/erlerobot/gym-gazebo#running-an-environment)solve the error.
I overlook that.

The simulator have launched,but turtlebot can't reinforcement learning.
It looks like it is slipping on Gazebo.And reward plot can't display.
Please heip me!! I'm sorry to bother you over and over.

taihirose, 2018-11-15:
Screenshot here.
[image]

taihirose, 2018-11-16:
OK. Thank you for answering the question. I'll do my best.

nzlz, 2018-11-21:
Hello again @taihirose . I have researched a bit about the display_plot error, related to the wrapper. You simply have to add the following import [here](https://github.com/erlerobot/gym-gazebo/blob/a545e7bbd3bcc5d0fd9c012d0bc42dab7324c579/examples/turtlebot/circuit2_turtlebot_lidar_dqn.py#L8):
`from gym import wrappers`. It will work this way :)

sayment, 2019-01-13:
I solved the same problem with the following command below

`cd ~/gym-gazebo
sudo -H pip install -e .`

nzlz, 2019-01-15:
Solved, closing.

closed 2019-01-15 by nzlz