erlerobot/gym-gazebo #185: How can I tell if I have reached the goal?

yangchen12, 2018-11-20:
In the GazeboCircuit2TurtlebotLidarNn,How can I tell if I have reached the goal?Anyone who can help me?Please

yangchen12, 2018-11-20:
> If I remember correctly, the reward function is designed in such a way the robot will learn to move as much as possible while keeping far away from the walls. Therefore there's no a real goal to reach in this scenario.

How should I set a goal in the scenario?Because in my experiment,I need a goal to judge if the algorithm is good.

yangchen12, 2018-11-20:
> If I remember correctly, the reward function is designed in such a way the robot will learn to move as much as possible while keeping far away from the walls. Therefore there's no a real goal to reach in this scenario.

can you give me a email?If I have questions,can I ask you？

pulver22, 2018-11-20:
@yangchen12 it depends which kind of target you want. It may be an object so you will use CNN for learning its shape as an important feature to track.

Otherwise you can use cartesian coordinates as target and calculate a reward based on the distance from your robot to this point.

PS: I don't know if I am entitled to share an email address on Github. It makes more sense you open issues here so many more people can help you and contribute to the general understanding.

yangchen12, 2018-11-21:
> @yangchen12 it depends which kind of target you want. It may be an object so you will use CNN for learning its shape as an important feature to track.
>
> Otherwise you can use cartesian coordinates as [...]

Thank you very much,if you have the relative codes,can you help me with the code？

nzlz, 2018-11-21:
Hello again @yangchen12 . You can modify the reward system as @pulver22 mentioned. 

A more simple approach would be to set a threshold for the obtained reward in the main algorithm.py script. The goal could be to obtain an average reward (with some minimum required value) during a few episodes higher than `somevalue`.

You could calculate the reward needed to complete one lap for example, or just set it manually. If the reward for completing a whole lap is achieved 20 consecutive times.. the goal is reached. This is just an example.

closed 2018-12-29 by nzlz