erlerobot/gym-gazebo #187: Turtlebot moving very slowly for qlearn

omkarpatil18, 2018-11-21:
Hi! I'm a newbie.
I've the following setup:
ROS Kinetic 
Gazebo 8.1.1
Python 3.5.2
OpenCV3, installed from sources for Python 3 
Open AI gym

The issue is that the robot is not being trained, and the script does not run completely. The simulation on gazebo shows the robot moving at excruciatingly slow speeds.
Screenshot of the markers I added in the code:
![qlearn_circuit2](https://user-images.githubusercontent.com/16060082/48872832-df6ab580-ee10-11e8-9709-d664ce97b724.png

On running circuit2_turtlebot_lidar_qlearn.py, the corresponding terminal looks like this:
[image]
[image]
So in effect, the code gets stuck on **observation = env.reset()** .  
The log file gives some errors:
[image]

I have looked at these issues:
[https://github.com/erlerobot/gym-gazebo/issues/60#issuecomment-348944142](url).
It suggests to comment line `# env.monitor.start(outdir, force=True, seed=None)`, which produced no effect.
[https://github.com/erlerobot/gym-gazebo/issues/146#issue-319126171](url). It suggests shifting to python2 for everything. 
Could you please suggest what could be done to resolve this issue?

nzlz, 2018-11-24:
You should definitely shift to python2. Python3 is not supported/tested.

pulver22, 2018-11-30:
I remembered that I had the same problem and I solved it recompiling everything with Python2.

But then, if you want to use any openai baseline for training, they are written in Python3 and it's really difficult to make everything work smoothly together. So I moved to Python3 and I was able to run Turtlebot example to work. 

I follow the guide reported [here](https://github.com/erlerobot/gym-gazebo/issues/90) and I didn't have any issue.

closed 2018-12-29 by nzlz